letscontrolit/ESPEasy #469: Converting alternative FW for Arilux led driver to plugin

Grovkillen, 2017-08-08:
I've talked to the developer behind the great FW for Arilux to join the ESP Easy community but he has no more time to give. His suggestion (also implied by myself) is that we try to merge his code into a plugin.

What do you all think?

The project page:
https://github.com/mertenats/Arilux_AL-LC0X

My short request:
https://github.com/mertenats/Arilux_AL-LC0X/issues/29#issuecomment-321041597

MatzeJoerling, 2017-08-30:
+1 
I would like to see more "Light" stuff in ESPEasy. Clock, Candle and Basic Neopixel is not really much.
Arilux looks really hard to merge, what about https://github.com/toblum/McLighting for a starting point,
I asked the author of McLighting - he aslo has no time to merge his project into ESPeasy. 

Someone has to volunteer...

regards Matze

papperone, 2017-09-22:
McLightining is heavily based on WS2812FX library, I'll take directly that library to a new plugin but it will be for 2.1 milestone... (does not seems too complex to implement!)
About Arilux I wish I could help but I need to have one controller for testing which I don't have right now...

nigelbe, 2017-09-22:
I would like to see a "user note" field where the user can put a short note about the particular node maybe up to 20 or 30 characters.

Grovkillen, 2017-09-22:
@papperone I have a controller and they are pretty nicely built. I wish I had two, to send you one :)

papperone, 2017-09-22:
thanks @Grovkillen but the issue is not only the controller but the fact that there are at least 7 or 8 versions and as well I would need proper LED Strip to test all functions (5ch up to "RGB+WW" I guess)...
for simple controllers/sensors I don't mind to order one from China and wait but in this case I think the Arilux are quite complex in terms of possible combinations of controllers and led strips... let's tackle first the WS2812FX wuich already has lots of possibilities!

PS: the Arilux is a standard RGB+WW similar to H801, so can't same plugin be used for both?

@nigelbe I think you wrote in the wrong thread or I can't understand your comment :)

Grovkillen, 2017-09-22:
Yeah I hear you @papperone , I got the simple RGB one ([found here](https://www.aliexpress.com/item/16Million-colors-Wifi-RGB-RGBW-led-controller-smartphone-control-music-and-timer-mode-magic-home-wifi/32755262961.html?spm=a2g0s.9042311.0.0.i2dbeF)). But they have multiple versions. Maybe just focusing on one would do it but the WS2812FX is a good candidate too I guess.

papperone, 2017-09-22:
still I think AriluX devices can be controlled by the H801 plugin, as they do the same job just maybe using different GPIOs... what do you think?

Grovkillen, 2017-09-22:
I will try the H801 plugin as soon as possible. Will investigate it more in detail.

tablatronix, 2017-10-08:
Why does this need a special library its just an esp with gpio out to 4 mosfets for rgbw pwm control.

Grovkillen, 2017-10-09:
The deal is that we want (I want) to use the possibility to control it using fairly extensive JSON messages as it is used with the FW I linked to in my first post. I might try to tinker with the board I have (the simplest [RGB controller](http://s.click.aliexpress.com/e/ybaU7YN)) and see what I find.

tablatronix, 2017-10-09:
I just ordered one,

Looking at the lib,  this should not be to hard to convert into a plugin, the arilux stuff is abstracted out from the library stuff in the ino, obviously you would not use any of the mqtt, telnet, json stuff from it, but use the espeasy way of doing that stuff, syslog, routing frontend, json will have to be rewritten to use the native methods, and hook into espeasy subscriptions or however you intend to control this.

You might even be able to keep the rf, ir control , but I think those would be better off hooked into espeasy as seperate devices.

Grovkillen, 2017-10-09:
Yes that was my thought as well (when I first posted it). Thanks for making me aware of this again! :+1: