letscontrolit/ESPEasy #470: "nan" value from sensor cause json parse error

daluma, 2017-08-10:
### Steps to reproduce
How can we trigger this problem?

NodeMCU v2 with DHT22
Configure as "usual" - is is working but sometimes reading errors from sensor
Error reading sensor cause "nan" value display which is OK but this results in a json parse error
Remove sensor to simulate the "nan" value

### Expected behavior
Even with one sensor is not providing proper values or nan value - it should be possible to get proper json page to collect the other information
 
### Actual behaviour
No valid json page is available - e.g. loading with php json information php json_decode is not possible for the failure case. Looking to json page on sensor reports a json problem too.

### System configuration

Hardware:
NodeMCU v2
DHT22

Software or git version:
v2.0.0-dev11

psy0rz, 2017-09-13:
this is indeed a bug. either it should send the `"nan"` as quoted string, or `null`.

psy0rz, 2018-01-12:
fixed (should be in tonight's builds)

closed 2018-01-12 by psy0rz