letscontrolit/ESPEasy #474: New Sonoff units need DOUT (not QIO) write mode when flashing

Grovkillen, 2017-08-13:
Should we make this part of the flash tool or just add a guide for how to do it with IDE?

https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=3386&start=10

TD-er, 2017-08-13:
What you describe [here](https://www.letscontrolit.com/forum/viewtopic.php?p=18182#p18182), it sounds to me like it is a compiler-setting?
So how could the flash-tool help here?
How much effort does it take to build extra images for the new types of Sonoff's?
I can imagine one different build isn't that much of an effort, but what when there are more options...

So if it is possible to fix such settings in the flasher, I think that's the best option. 
And ofcourse describing it somewhere is always a good thing :)

Grovkillen, 2017-08-13:
Yeah we get a lot of questions since Sonoff is a big user device. We definitely need to document it. But if it is possible to fix with the tools it would be great. It's more of a "heads up" that some hardware that is officially supported had started to behave differently with newer batches. :)

ShardanX, 2017-08-14:
The esptool.py written for linux is able to set DIO mode on the fly via command line, i don't know if the esptool.exe for windows can do. It is not really a compiler option, as far as i know - if i remember right the compiler writes an info about flashing mode into the binary's header. I might err with that.

Anyways it should be documented in the wiki that at least some versions of the sonoff need DIO mode - my older ones run without this setting.
If the windows esptool.exe can be set within the command line, it shouldn't be too difficult to change the tool to do the job.

TD-er, 2017-08-16:
As I read on the [Tasmota wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips), all ESP8266 and ESP8265 devices support DOUT and therefore Tasmota is changing to only use DOUT for all devices.

So given that reasoning, I would suggest to do the same and make sure all flashing actions are done using DOUT.
N.B. I suspect also for OTA updating the firmware, the DOUT must be used in the firmware itself.

For my own reference (since these modi were new to me):
* DIO - SPI host uses the "Dual I/O Fast Read" command (BBh). Two SPI pins are used to write the flash address part of the command, and to read flash data out. Therefore these phases need half the clock cycles compared to standard SPI.
* DOUT - SPI host uses the "Dual Output Fast Read" command (3Bh). Two SPI pins are used to read flash data out. Slightly slower than DIO, because the address is written via the single MOSI data pin.
* QIO - SPI host uses the "Quad I/O Fast Read" command (EBh). Four SPI pins are used to write the flash address part of the command, and to read flash data out. Therefore these phases need a quarter the clock cycles compared to standard SPI.
* QOUT - SPI host uses the "Quad Output Fast Read" command (6Bh). Four SPI pins are used to read the flash data out. Slightly slower than QIO, because the address is written via the single MOSI data pin.

In terms of performance: QIO > QOUT > DIO > DOUT    ([source](https://www.esp32.com/viewtopic.php?p=5523&sid=c5b0928b71adfd40018be2ddab595e00#p5523))

psy0rz, 2017-09-14:
I think we already have the precompiled ESP8285 version, we created that for the sonof devices, right?

https://github.com/letscontrolit/ESPEasy#firmware-binary-releases

So its only a matter of documenting, right? (and maybe adjusting the flash tool)

Grovkillen, 2017-11-27:
I will try to fix the documentation regarding this during this weekend.

Grovkillen, 2017-12-02:
https://www.letscontrolit.com/wiki/index.php/ESP_Hardware

I added information that newer Sonoff might use 8285 (not sure if it is entirely correct but since ESP8266 and 8285 are basically the same it's safe to flash an 1MB, or bigger, with the ESP8285 bin file).

closed 2017-12-02 by Grovkillen

emeyedeejay, 2018-03-08:
This particular issue/change in the requirement of the flash mode personally cost me 2 days until I finally fathomed it out by making a link between the various commands I had seen to restore a backed-up flash and a post about integrating the ESP8266 (on a SonoFf WiFi switch) into the Arduino IDE.

That happened to mention the flash mode should be set to DOUT but the commands I was trying all had QIO in them. Once I tried DOUT, all worked perfectly.

TD-er, 2018-03-08:
A few days ago, I changed all 1M builds to DOUT mode. So for newer builds it should now work.
Can you confirm it does work as expected on newer builds (for 1M flash chips)?

emeyedeejay, 2018-03-08:
I was actually trying to get tasmota firmware going on a SonOff wifi switch - sorry, I crossposted here (although it is the same issue, just not the same firmware).

jobst, 2018-03-17:
I received a bunch of Sonoff Basic swtiches a few days back. I downloaded the EspEasy stuff from here: http://www.letscontrolit.com/downloads/ESPEasy_R120.zip, setup the Usb-Serial connection, put the sonoff into programming mode and flashed the device. Nothing worked although the uploader told me that flashing was successful. 5 hours of research later and lots of people providing different/confusing/opposing advice I downloaded Arduino 1.8.3 (after people suggested that 1.8.6 is a no no), loaded a led switching program and used DOUT - and it worked immediately. So I copied the ESP binary to my Linux box and used the esptool.py. This tool allows to use DOUT as programming mode - again it worked immediately.
It would be good if the esptool.exe provided in the ZIP file allowed setting of DOUT programming mode as the latest Sonoff must use DOUT.

Grovkillen, 2018-03-17:
We provide DOUT bin files already plus that you can use my Flasher if you're on Windows.

eagle696, 2018-04-05:
I have the same problem with the new sonoff EEPROM Devices. I use Arduino IDE with OTA, flashmode DOUT works when i flash over cable.
 OTA uploads sucessful,  but programm is not running. 
What do I have to do to flash the new sonoff devices over Arduino OTA? 
Thanks for your help....

Oxyandy, 2018-04-05:
@eagle696  This help ?
https://github.com/letscontrolit/ESPEasy/issues/1190#issuecomment-377054077

uzi18, 2018-04-06:
Try to get terminal log for us.

eagle696, 2018-04-10:
Thanks for your answers...but I am still standing on the line. 
I never used espEasy, or any".bin" file.
I flashed all my esp8266 devices with the normal Arduino IDE 1.8.4 with OTA Function (just hat to change the old EEPROM to 4M) 
@ uzi18  .. what do you mean with terminal log in the arduio IDE? The Arduino IDE just gives me the information: 
Uploading..................................................................................................upload done
And then I have to go back to the cable to get the new sonoff to work again. (DOUT is set in the Arduino IDE) 

Can you help me to tep from the line, please?

eagle696, 2018-04-10:
In case that there are some other options, I only used my own Arduiono .ino files to flash

Grovkillen, 2018-04-10:
Can't you try an official bin file to start with? It's kinda hard to help
the custom builders with no files to look at.