letscontrolit/ESPEasy #477: Deep Sleep delay over 1h 11m (to reduce battery use)

k-sartor, 2017-08-15:
As discussed in this thread:
https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3165&hilit=deepsleep&start=10

There is a ticket to ask an enhancement to increase the deep sleep delay over its limit of 1h 11m
Indeed, if a time over about 4200s is put on ESPEasy, the deep sleep occurs each 1h and 11m.

This behaviour is due to the deepsleep counter is a 32 bits counter.

A workaround is use the RTC memory to check is a higher deep sleep delay is over or not. Therefore the wakeup will be done only this higher deep sleep delay is over

Grovkillen, 2017-08-16:
An implementation of the use of a RTC (DS3231, which has i2c) upon wake would be great. It could then check once every 1...4200s (set by the user) and if the clock is close to the trigger set time it should continue to wake, if not go back to deep sleep.

Another way that is already possible is to have a external trigger timer which sends a signal to the deepsleep wake GPIO using a device like this (P1-3 would work as intended, I believe):
[Timer device](https://www.aliexpress.com/item/1pc-12V-LED-Home-Automation-Delay-Timer-Control-Switch-Relay-Module-Digital-display/32666199799.html?spm=2114.10010108.1000014.10.4b24ae2b1FHow0&traffic_analysisId=recommend_3035_null_null_null&scm=1007.13338.80878.000000000000000&pvid=492bd458-59bb-444a-97c4-1f51c11c1b6f&tpp=1)

This device can operate a countdown of maximum 999 minutes which is a little more than 16.5 hours. If it is set to the P1-3 setting the device will activate the relay for X seconds/minutes once the Y (count down) is reaching zero. Then restart the loop.

psy0rz, 2017-09-14:
we shouldn't need an rtc for it, like i stated on that forum post 

"Yeah we could even make it easier for the user: Just let the user specify the actual sleep delay they want. If its longer than the max sleep time of the ESP, then wake up every ~4 hours until we can do the final sleep after which we do our task."

will implement this after 2.0.0

enesbcs, 2018-09-09:
DS3231 beta plugin:
https://github.com/enesbcs/ESPEasyPluginPlayground/blob/master/_P157_DS3231.ino

TD-er, 2019-10-27:
It has been added.
Needs a newer core library (core 2.5.x or newer) to function.

closed 2019-10-27 by TD-er