letscontrolit/ESPEasy #484: Possibility to turn on/off wifi for stand-alone-units

Grovkillen, 2017-08-21:
It would be a great feature to have the possibility to turn off wifi for stand-alone-units that are meant to be used offline. A possible way of turning an offline unit back online would be to integrate a wifiOn command which could be connected to a toggle switch.

**wifiOff** will turn the wifi off and not have the unit opening up the AP mode
**wifiOn** will turn the wifi on and trying to reconnect to the network, if no access is available the AP mode will be triggered.

TD-er, 2017-08-22:
Just curious... How do you give the command to turn the WiFi on again?
A button-command or something?

Grovkillen, 2017-08-22:
Yes, I might not have explained myself clear but I stated: "wifiOn command which could be connected to a toggle switch."

You may always connect to it using serial commands but for easier maintenance a "wifi toggle switch" would be the solution.

psy0rz, 2017-09-14:
The WifiDisconnect and WifiConnect commands might do this? Or can we also actually turn of part of the wifi-hardware?

Grovkillen, 2017-09-18:
I will have to try and see if this is what I'm aiming for. I just thought it would be possible to turn off the hardware, I might be wrong.

mikeluyten, 2017-10-16:
Would be good if the DS1307 RTC module could then be used to keep the time if power is lost/interrupted?

enesbcs, 2017-12-07:
Modemsleep like this?  :)
https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=2846#p21370

Grovkillen, 2018-03-05:
I think this issue/feature request would also make it possible to operate switches on units that has lost wifi connection. For example my garage door opener will not work if my AP is down due to what ever reason.

Expected behavior on units with input possibilities (i.e. switches etc.) is that they should work even if wifi is lost.

TD-er, 2018-03-05:
Agreed, the ESP should be able to operate as autonomous as possible.
That's also the reason for this issue: #1011 
Internal logging allows for local buffering of samples to be flushed later (if controller allows to send past values with a timestamp)
And rules should still continue when not connected to the network.

mikeluyten, 2018-03-05:
I found on my units that when wifi goes down it goes into a cycle of trying to connect to wifi over and over again and while trying to connect, no other commands run, not even an LCD screen will update. So the issue your having is more of a load/resourcing issue while its trying to reconnect.

Maybe if there was a wifi connection/reconnection options like:

Manual (stay dead)
Automatic (keep trying to reconnect) - current behaviour
Automatic (On wifi failure, timeout of x min before retrying)

Manual you could toggle wifidisconnect and wificonnect via rules?

Cheers,

Michael Luyten
0423 380 911

TD-er, 2018-03-05:
@mikeluyten I have changed some of this need to retry in the last few builds.
Did you experience this on the last few builds also?

There is still a lot more room for improvement on this.

mikeluyten, 2018-03-05:
I am running Build V2.0-20180213

Cheers,

Michael Luyten
0423 380 911

TD-er, 2018-03-05:
Would you be so kind to try a newer build?
My approach is to not try to do all in one cycle of the main loop function.
So if connection fails, try again later, not now and if it keeps failing, increase the retry interval.

Also DNS resolve does almost halt the ESP (if request fails), so use cached IPs for controllers. When the cached value fails, try again with a DNS lookup.

mikeluyten, 2018-03-05:
@TD-er Trying to update now, not having a great time, getting Update Failed! on all my ESP's and it fully wiped the config flash. (I probably should have backed it up... but didnt...) Trying over usb/serial now
V2.0-20180305

Grovkillen, 2018-03-05:
Ooops... missed this [information](https://www.letscontrolit.com/wiki/index.php/ESPEasy#2.0.2B_firmware_on_1MB_modules)?

mikeluyten, 2018-03-05:
Sure did, I swear I only just recently updated some OTA... Just realised, the one I did OTA has 16MB flash... which unfortunately died last week, forcing me to use some older 1MB versions...

FEATURE REQUEST don't allow it to upload/try to flash when there isn't enough space? like a warning or something saying a OTA up-loader is required... for cowboys such as my self.

Grovkillen, 2018-03-05:
#708 is discussing exactly that issue.