letscontrolit/ESPEasy #486: Bug in pulse counter plugin (reported on forum)

Grovkillen, 2017-08-24:
```
          case 3:
          {
            event->sensorType = SENSOR_TYPE_DUAL;
            UserVar[event->BaseVarIndex] = Plugin_003_pulseCounter[event->TaskIndex];
            UserVar[event->BaseVarIndex+1] = Plugin_003_pulseTotalCounter[event->TaskIndex];
            break;
          }
```
Should be

```
          case 3:
          {
            event->sensorType = SENSOR_TYPE_DUAL;
            UserVar[event->BaseVarIndex] = Plugin_003_pulseCounter[event->TaskIndex];
            UserVar[event->BaseVarIndex+1] = Plugin_003_pulseTotalCounter[event->TaskIndex];
            break;
          }
```

https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3476#p18576

uzi18, 2017-08-24:
Is there any difference ?

Grovkillen, 2017-08-24:
Sorry, copied that one twice. The first post is now updated.

ShardanX, 2017-08-24:
Might this be the reason for the issue in #454 ?

closed 2017-09-14 by psy0rz