letscontrolit/ESPEasy #487: PMSx003 fields, GPIO description

ShardanX, 2017-08-24:
### Steps to reproduce
How can we trigger this problem?
Connect a PMSx003 as described and check values.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Definitely

### Problem
Plugin gives values in particles counted/0,1l.
Usually (at least here) are values in µg/m³,
There is no formula field in the plugin for a conversion formula,
decimals are missing either.

The description "1st GPIO... 3rd GPIO" should be extended with
with the pin descriptions of the sensor (Rx, Tx, Reset) to make use easier.
Took me a while to find out....

### System configuration

Hardware:
WeMOS D1 Mini with PMS5003

Software or git version: 
Git from August 18., 2017.

ShardanX, 2017-08-28:
I've checked the source code of P_053 meanwhile.... it seems the given values are the µg/m³ under atmospheric conditions, as far as i can see as a non-programmer.
Anyways the values are extremely high for a home office:
```
PMSx003 : Packet available
PMSx003 : pm1.0=490, pm2.5=1253, pm10=1575, pm1.0a=326, pm2.5a=834, pm10a=1049
PMSx003 : count/0.1L : 0.3um=13601, 0.5um=26006, 1.0um=13401, 2.5um=1458, 5.0um=375, 10um=7
```
from the serial log.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er