letscontrolit/ESPEasy #491: v2.0.0-dev11: two INA219 Devices get theast submitted i2c address will be used for both devices

ThatRed, 2017-08-25:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
Tell us what should happen?

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:

Software or git version:

Grovkillen, 2017-09-02:
More info here:
https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=3479#p18796

psy0rz, 2017-09-15:
yep its a bug. the global variable should be stored in taskvalue arrays.

Grovkillen, 2017-09-22:
@psy0rz add this to 2.0.0 milestone

TD-er, 2017-09-27:
I think the same applies for the BME280/BMP280 plugins.
But it is more likely to use multiple power-sensors at the same controller, compared to how likely it is to use several temperature sensors at the same controller.
i2c address is also stored in a global variable, but the boolean flags for init and calibration are stored in an array as I think it should be.

Is it safe to assume also these parameters of the INA219 may be affected when using more sensors with different ranges set?
```
uint32_t ina219_calValue;
// The following multipliers are used to convert raw current and power
// values to mA and mW, taking into account the current config settings
uint32_t ina219_currentDivider_mA;

```

TD-er, 2017-09-28:
I just made some changes and created a pull request, as I think it should fix this issue.
I do not have this sensor, so please test with real hardware.

psy0rz, 2017-09-30:
@ThatRed can you test it?

also try powering it off and on when it works, to see if all important data is actually stored in flash. :)

psy0rz, 2017-10-12:
ordered hardware, will test later myself.

TD-er, 2017-10-13:
Today I received 3 of these sensors and tested them.
See pull request #540
Only minor changes in comments and log-entries, but is working fine.

Just a note on testing the sensors.
The Vin- and Vin+ pins on both sides are connected to the 0.1 Ohm shunt. So do not connect a power source to both these pins, like I did. (oops)
Voltage is just measured between GND and Vin-
Current is measured between Vin- and Vin+

mvdbro, 2017-11-05:
fixed in PR #519

closed 2017-11-05 by mvdbro