letscontrolit/ESPEasy #496: Feature request - rpm task

themanfrommoon, 2017-08-28:
Hello,

it would be nice to have a rpm task (or tachometer or speedmeter)
I tried some things here:
https://www.letscontrolit.com/forum/viewtopic.php?t=3476
But in the end it does not really work very accurate.

It would be great to count the pulses for 60 seconds, then use this value as output and reset the counter to zero to start the procedure again.
One rpm value every 60 seconds (or adjustable period) would be sufficient.

for example:
after 60 seconds: output is "3994" rpm
after 120 seconds: output is "3995" rpm
after 180 seconds: output is "3991" rpm
after 240 seconds: output is "3993" rpm
and so on

Maybe this should be a new task or an enhancement of the existing counter task.

Thanks for your support and this great software!

best regards,
Chris

beicnet, 2017-08-28:
Dear @themanfrommoon for Feature request, please post it on the ESPEasy forum!

themanfrommoon, 2017-08-28:
Hello beicet,

I've done it the same way someone else descriped it in the forum: 
https://www.letscontrolit.com/forum/viewtopic.php?t=3303
But this seems not to be right?!

Where should I post it exactly?
Is there a special area or a special thread where to post it?
How should it be posted?
Maybe with a special topic?

Thanks and best regards,
Chris

themanfrommoon, 2017-08-28:
I tried it as good as I can and some people tried to help me.
But in the end it does not work (very accurate).
For my understanding "ESPEasy" is made for simple end users to make a little bit complicated stuff much easier.
For that reason: Yes, I want a RPM Task/Feature.
I guess it makes a lot of sense.

So how should I proceed exactly with the feature request?

Thanks and best regards,
Chris

beicnet, 2017-08-28:
Post it [here](https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=731&start=180) like "RPM Task" plugin...

beicnet, 2017-08-28:
_"For my understanding "ESPEasy" is made for simple end users to make a little bit complicated stuff much easier."_

But, it's limited...

themanfrommoon, 2017-08-28:
Done:
https://www.letscontrolit.com/forum/viewtopic.php?&t=731&start=190

beicnet, 2017-08-28:
And now we wait for miracles! ;)

closed 2017-09-15 by psy0rz