letscontrolit/ESPEasy #498: Refuse to connect Wifi when SSID with Quote character inside

thiboot, 2017-08-30:
### Steps to reproduce
Try to connect to a wifi with a quote "'" in the SSID
If i change SSID without quote it worked
Hypothesis: SSID field is not escaped

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
YES

### Expected behavior
It should connect

### Actual behavior
Refuse to connect to Wifi

### System configuration

Hardware:
Node MCU v3

Software or git version:
ESPEasy_v2.0.0-dev11_normal_8285.bin

psy0rz, 2017-09-15:
i didnt know this was actually supported in wifi.

probably a escaping bug in our stuff.

thiboot, 2017-09-17:
Yes it is upported, you can even put white space and Emojis 😂

802.11 2016 Standards:
http://standards.ieee.org/getieee802/download/802.11-2016.pdf

**A valid SSID is 0-32 octets with arbitrary contents**. A 0-length SSID indicates the wildcard SSID (in probe request frames for instance).

There's no character set associated with the SSID - a 32-byte string of NUL-bytes is a valid SSID.

This implies:
- you should never use normal string functions when manipulating generic SSIDs (strcpy() and friends).
- you should not assume that the SSID is printable when, for instance, logging it to disk

(https://stackoverflow.com/a/5017144)

psy0rz, 2017-09-18:
ah i think we use it as string so far.

Grovkillen, 2018-03-16:
Is this something we intend to fix to 2.0 or should we bump it to 2.1?

closed 2019-10-11 by Grovkillen