letscontrolit/ESPEasy #499: SDS011 lifetime problem

ShardanX, 2017-09-02:
### Steps to reproduce
How can we trigger this problem?
Use sensor as described...
Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
yes
### Expected behavior
Tell us what should happen?
Sensor should be put to sleep for "Delay" time.
### Actual behavior
Tell us what happens instead?
Sensor runs permanently
### System configuration
Hardware:
WeMos D1 with sensor
Software or git version:
Latest from github

It is the same problem as described for the PMSx003 in #487 
Due to datasheet the laser diode inside the sensor has a lifetime of around 8000 hours.
Permanently running the sensor will reduce the lifetime to below one year.

The protocoll data sheet of the sds-sensors says it can be put to sleep and wake up by sending code over the serial port.

I suggest to change the plugin to reading until a reasonable value (checksum ok) arrives then put the
sensor to sleep for the time set in the "Delay" field.

TD-er, 2017-09-28:
Bugs like these are just a perfect excuse to buy such sensors :)
Mine has just arrived, so I will have a look at it later this week (I hope).

TD-er, 2017-11-05:
I was just looking into the source code of this sensor, since it is already on my desk for about 6 weeks.

Just looking into the history of this sensor: https://github.com/letscontrolit/ESPEasy/pull/451/files
The old version of the plugin used this library: https://github.com/ricki-z/SDS011/blob/master/SDS011.cpp
That old library has some sleep command to put the sensor into sleep-mode.
The new library used ([CjkSDS011](https://github.com/jkDesignDE/ESPEasy/commit/643dfc160271ac104bb00587e187b8e50bfa85fc)) does not implement such a sleep function.

@jkDesignDE 
Is there a reason the sleep option was taken out? Perhaps not working like expected?

TD-er, 2017-11-05:
Just looked into the [Laser Dust Sensor Protocol v1.3](https://cdn.sparkfun.com/assets/parts/1/2/2/7/5/Laser_Dust_Sensor_Control_Protocol_V1.3.pdf)
Section "5) Set working period" (page 8/9) does  exactly what is needed here.
Only thing is the ReadAverage function does currently not allow for readings with 0 measured values.
So either that function could return a previous value, or the plugin should handle the NaN values.

TD-er, 2017-12-31:
I just added a pull request for this feature.
When accepted, this issue can be closed.

closed 2018-01-02 by psy0rz

sisamiwe, 2018-04-09:
Hi,
I'm using EspEasy Mega release dates 2018-04-07 on an Wemos D1 mini connected with BME280 and SDS011.

To enhance the SDS011 lifetime I wanna use the sleep mode as it is implemented in the plugin to have just one reading per 30min. The sensor is active for 30 sec (fan is on) then it sleeps for 30 sec and starts again. 

So I configured the SDS011 as the following:
 - GPIO ← TX: GPIO-4 (D2)
 - GPIO ⇢ RX (optional): GPIO-16 (D0)
 - Sleep Time: 30 min
 - Data Acquisition: Delay: 600s

What I'm doing wrong?