letscontrolit/ESPEasy #502: Notifications does not work properly

ghtester, 2017-09-09:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?

Go to Notifications tab
Edit Nr1 - create an E-mail notification 1
Edit Nr2 - create an E-mail notification 2 with different parameters
Enable both notifications

Go to Tools tab 
Submit Command: notify 1 
Submit Command: notify 2

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
yes

### Expected behavior
Tell us what should happen?

Every notification should be independent on another one and all should work as expected. 

### Actual behavior
Tell us what happens instead?

In both cases (submit command notify 1 , submit command notify 2 ) just Notification Nr1 is performed.

I have also encountered some strange behaviour when all three notifications were entered and edited several times (both E-mail and Buzzer).  Their settings got screwed up together and I had to reset the configuration to default. 

### System configuration

Hardware:

ESP-12E

Software or git version:

v2.0.0-dev11
Build:  20000  - Mega
Plugin sets: | [Normal] [Testing]

ghtester, 2017-09-23:
The same issue in v2.0.0-dev12 version. On another device (NodeMCU Amica) v2.0.0-dev12, Plugin sets: |[Normal] [Testing], the Nr1 notification is an E-mail which works fine, the Nr2 notification configured as Buzzer which does nothing, just responds with OK.

psy0rz, 2017-10-12:
well.. that took 2 hours :)

closed 2017-10-12 by psy0rz

Grovkillen, 2017-10-13:
Good job @psy0rz

ghtester, 2017-10-13:
Thanks for your great work! :-)

commit 60f5bdb0 (2017-10-12): cleaned up notification stuff. fixed #502