letscontrolit/ESPEasy #503: Rules comparison reversed

gaetanlord, 2017-09-10:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce

Configure a ds18b20 device
I have a led connected on GPIO2

This command shutdown the led
http://192.168.1.238/control?cmd=GPIO,2,0

Define the following rules
On Rules#Timer=1 do
  if [ds18b20#Temperature]>21
     gpio,2,0
  else
    gpio,2,1
  endif
  timerSet,1,5
endon

Problem persist after a reboot

### Expected behavior
I would expect to enable GPIO2 (led ON) when temperature goes over 21 and actually it is the opposite.  The led is ON when temperature is below 21

### Actual behavior

### System configuration

Hardware: NODEMCU

Software or git version: 
Build:	20000 - Mega
GIT version:	v2.0.0-dev11

uzi18, 2017-09-10:
Did you connected anode of led to +3v and kathode to gpio?  If yes -  it
will work reversed.

papperone, 2017-09-10:
where is the problem? your rules says if temperature is greater than 21 (>21) then GPIO,2,0 (LOW = OFF) else GPIO,2,1 (HIGH = ON)
PS: this is not the right place to get thi kind of support, you shoudl have posted in the forum as alerted!

gaetanlord, 2017-09-10:
Shame on me,  your right, I  should have ask a question than filing a bug. Sorry, I'm learning this great tool

gaetanlord, 2017-09-10:
No bug I  rules

Grovkillen, 2017-09-11:
So please close it.

closed 2017-09-11 by gaetanlord