letscontrolit/ESPEasy #505: Ser2Net: periodically send message over UART and confuse serial device

crememanfred, 2017-09-14:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
**Use UART as usual.**

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
**YES**

### Expected behavior
Tell us what should happen?
**Checkbox in Ser2Net plugin (or in Config) for disable periodically send message over UART.**

### Actual behavior
Tell us what happens instead?
**Approx every 30 seconds message "WP: XX Uptime... XX FreeMem..." is send by UART and confuse my serial connection device.**

### System configuration

Hardware:
**NodeMCU V1.0**

Software or git version:
**ESPEasy_v2.0.0-dev11**

psy0rz, 2017-09-14:
you can solve this by setting the serial log level to 0 i think?

closed 2017-09-14 by psy0rz

crememanfred, 2017-09-15:
Where can I find this setting?

beicnet, 2017-09-15:
@crememanfred in the Advanced tab, I think!

crememanfred, 2017-09-18:
Thank you guys!

beicnet, 2017-09-18:
@crememanfred  Don't mention it! ;)