letscontrolit/ESPEasy #511: Switch input not keeping state

lggrizendi, 2017-09-19:
### Steps to reproduce
How can we trigger this problem?
I using a relay board in GPIO-14 and  power  off and on the esp8266, i use to set the fridge on and after a power loss is the system is not on, i have to set the state  i put after the reboot.
Internal PullUp: off
Send Boot state:off

Does the problem persist after powering off and on? (just resetting isn't enough sometimes)
yes

### Expected behavior
Tell us what should happen?
The relay should be powering on  

### Actual behavior
Tell us what happens instead?
The relay is  powering off, the state is 0 but the esp8266 is not keeping  the signal after reboot or power off and on 

### System configuration

Hardware: esp8266 nodemcu esp-12e

Software or git version:ESPEasy_v2.0.0-dev11_normal_4096

sorry for my bad english

lggrizendi, 2017-10-09:
I manage to resolve this issue. 
On system rules a add: 

on System#Boot do 
    gpio,14,status
endon

psy0rz, 2017-10-12:
This is currently not possible: switch states are forgotten if you restart or poweroff/on the ESP.

Maybe you can work around it by making a correct http post to the /hardware page and change the boot-state of a pin.

Also the ESP should keep the pinstate in deepsleep mode. (not sure if this still works/ever worked).

The stuff you want requires some new functionality that stores the device task data to persistent storage. We can do that, if more people need it.

If someone needs this leave a comment and vote up. For now i'm closing it.

closed 2017-10-12 by psy0rz

virenderbhanot, 2017-12-31:
This feature is indeed required.

psy0rz, 2018-01-02:
People watching this: Vote up the features or bugfixes you want like to see fixed first. Use the thumbs UP arrow for that on the first comment. This way we can determine where to start, because we have LOTS of requests. :)