letscontrolit/ESPEasy #512: Restore / upload config.dat and rules.txt using url in script

kuipers250, 2017-09-20:
I am using a script to backup the configuration of the ESP's in my network and create a daily zip file of it.
/config.dat /rules1.txt etc.
But there is now way to upload the file back to the esp using a script with simple url commands.
Maybe this can be implemented in a future release?

I need to have the sunrise/sunset time in a rule. Is there another way?

Grovkillen, 2017-09-20:
You could publish the sunrise/sunset time using HTTP or MQTT. But the request might still be valid in other cases (batch update the ESP swarm etc.).

kuipers250, 2017-09-20:
Please tell me how to publish a variable using HTTP. I don't use MQTT.

Grovkillen, 2017-09-20:
You create an event with given time (inside rules) and call it using the event command:

http://<espeasyip>/control?cmd=event,sr0615

Which in this example would set the SunRise to 6:15. Depending on your wanted resolution you need to create an event for each one of them.

A simpler way would be to use MQTT in conjunction with MQTT Import which would let you send the hour as one package and the minute as one. Then using rules and Dummy device that would be joined as an time.

Grovkillen, 2018-04-11:
I close this issue, reopen if still valid (sorry for misunderstanding if that's the case)

closed 2018-04-11 by Grovkillen