letscontrolit/ESPEasy #513: Parse the JSON message from MQTT "domoticz/out"

tuxmartin, 2017-09-21:
Domoticz send this json for relay change state:

```
$ mosquitto_sub -v -h 10.123.1.18 -p 1883 -t '#'
domoticz/out {
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014052",
   "idx" : 2,
   "name" : "Sonoff_1_internal_relay",
   "nvalue" : 1,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "On/Off",
   "unit" : 1
}
```
but ESP Easy doesn't parse the JSON message: https://www.letscontrolit.com/forum/viewtopic.php?p=19197#p19197

URL control works `http://10.123.1.188/control?cmd=GPIO,12,1`:
```
{
"log": "GPIO 12 Set to 1",
"plugin": 1,
"pin": 12,
"mode": "output",
"state": 1
}
```
but is not possible to use DHCP for ESP8266.

Please add MQTT "domoticz/out" support like in https://github.com/arendst/Sonoff-Tasmota

Grovkillen, 2017-10-06:
Maybe a future JSON parser would do the trick? https://github.com/bblanchon/ArduinoJson

I put this on "Future" and "Enhancement"

Grovkillen, 2017-10-06:
#527 is related (closed)

tuxmartin, 2017-10-06:
> "description" : "", <------ removing this line from the message to ESPeasy solves the problem.
> https://github.com/letscontrolit/ESPEasy/issues/527

mvdbro, 2017-10-08:
It still works here without modification. There could be issues with the buffer size (the library will just ignore those messages)
Are u using the patched pubsubclient library from our github?

tuxmartin, 2017-10-08:
I tried to publish only `idx`, `nvalue` and `svalue`:
```
$ mosquitto_pub -h 10.123.1.18 -t domoticz/out -m '{"idx": 2, "nvalue": 1, "svalue1": "1"}'
```

still not working :-(

tuxmartin, 2017-10-09:
This works for me: https://www.letscontrolit.com/forum/viewtopic.php?p=19717#p19717

closed 2018-03-25 by Grovkillen