letscontrolit/ESPEasy #514: Publish command in rules not work as expected with Serial Server (Ser2Net) plugin

agsochi, 2017-09-22:
Hi!
I use Serial Server (Ser2Net) plugin. 
[image]
 Serial Server works - i see messages from serial port:
```
1389045 : EVENT: !RFLink#
1389052 : ACT : publish /test,
1389383 : EVENT: !RFLink#eeloq;ID=9ac2;SWITCH=01;CMD=ON;BAT=OK;
1389389 : ACT : publish /test,
1403244 : EVENT: !RFLink#Keeloq;ID=9ac2;SWITCH=02;CMD=ON;BAT=OK;
1403250 : ACT : publish /test,
1403909 : EVENT: !RFLink#
1403915 : ACT : publish /test,
1404207 : EVENT: !RFLink#eeloq;ID=9ac2;SWITCH=08;CMD=ON;BAT=OK;
1404213 : ACT : publish /test,
```
My rules (if i use 'on rflink do' - rule do not work):
```
on [rflink] do
publish /test,[rflink]
endon
```
 In MQTT i see only:
```
/test (null)
/test (null)
/test (null)
/test (null)
```

**Issue - no `rflink` value published.**

Sorry for question, i didn't know where i can get help - How to send command back to 'Serial Server' from 'Rules'?

papperone, 2017-09-22:
first of all this is a message more for the forum than for the GITHub as there's yet no evidence this is a bug, then I believe you are not using the correct syntax as square bracket shouldn't be used in the ON command as far as I remember....

agsochi, 2017-09-22:
I try all cases with square bracket and without it. Without square bracket - no event at all.

Grovkillen, 2017-09-22:
Shouldn't the rule be:

`on rflink# do
publish /test,[rflink]
endon`

agsochi, 2017-09-22:
> on rflink# do
publish /test,[rflink#]
endon

No, this rule do not work too.

papperone, 2017-09-23:
I think your event name includes the esclamation mark "!" as in all my events that's not there (maybe a "feature" of this plugin); as well some events looks like weird ending with a pound "#" without the second argument... I am convinced your issue is specific to the events generated by this plugin, have you tried to open this thread in the forum?
GitHub shoudl be only for known bugs in the code, while this looks like more help to make it work!

psy0rz, 2017-09-23:
please re-open if someones sure this is a bug. thanks :)

closed 2017-09-23 by psy0rz

agsochi, 2017-09-25:
I change name to`rflinkraw`, and i found that right rule trigger name is `!rflink` :
```
on !rflink do
publish /test1,ok
publish /test1a,[!rflink#rflinkraw]
publish /test1b,[!rflinkraw]
publish /test1c,[rflinkraw#]
publish /test1d,[!rflinkraw#]
publish /test1e,!rflinkraw#
endon
```
But still no luck in publishing.