letscontrolit/ESPEasy #515: rewrite of PLUGIN_WRITE handling as not working with multiple device of same kind

papperone, 2017-09-22:
### Steps to reproduce
Part of the CORE design of ESPEasy; this is more to raise a concern to analyze if this limit can be somehow overcome in the future releases...

### Expected behavior
the PLUGIN_WRITE functions should be able to access device data and as well be able to address command to a specific device (in case of multiple devices configured)

### Actual behavior
the PLUGIN_WRITE functions cannot discriminate on which device the command is referred to, and due to this the device data/parameters can't be accessed; on top of this commands are executed for all devices of same type (e.g. attaching 2 NeoPixel strings it's impossible to control them separately via the specific command)

### System configuration
ANY

psy0rz, 2017-09-23:
i have to look into it, might bump it to beyond 2.1.0 .

mvdbro, 2017-10-08:
Not entirely true if you look at plugins like the P009_MCP, P007_PCF8591, P019_PCF857 that can all handle multiple devices. It's entirely up to the plugin if it can handle multiple devices or not.

I don't see why this is considered a bug. Just a design limitation on some plugins.

The PLUGIN_WRITE can determine task configuration if it's command syntax has any reference to it.
The ESP Easy core does not enforce any syntax rules on the plugins. A bit dangerous but also very flexible.

papperone, 2017-10-08:
Let me explain better as I maybe sum up 2 different issues that can have different impact in how/when/if to be solved :)
1) currently the PLUGIN_WRITE cannot access the "EventStructure" of the device as this is not part of the design but I do believe this is a bug to be fixed as this limits a lot plugin capabilities not being able to access the configuration details of the plugin (e.g. NeoPixel cannot determine in PLUGIN_WRITE how meny leds have been configured and I submit a "patch" that saves this value in a variable to make it working properly)
2) As well the PLUGIN_WRITE system is IMHO to be reviewed as it's true you can have multiple devices using same plugin but as it is designed you cannot send a command to a specific device if multiples are defined but only the frist device will get/execute the command also with the limits of point 1)
Again I take NeoPixels as an example, if you define 2 separate strips of neopixels, while perfectly legit and useful in many situations, you won't be able to control them properly as PLUGIN_WRITE won't have a clue about each device parameters (again number of pixels of a specific device) and you cannot send a command to a specific device but only the first one will react and execute

Plugins mentioned they indeed works but only because either they do not have any device specific parameter to be set (so no need to access them!) or they do not have PLUGIN_WRITE enabled...

papperone, 2017-10-08:
PS: you cannot setup two P009_MCP devices as only the first one will be able to exectute PLUGIN_WRITE for the limits we are talking about, so definitely this plugin while it could (and should!) cannot be used for multiple devices

mvdbro, 2017-10-08:
I developed both the MCP and PCF plugins and believe me or not, they DO work properly with multiple devices. They have been build, tested and verified with that requirement.
And you do not even have to setup these devices for output.

Reason:
The syntax as defined within the plugin uses an addressing mechanisme that takes care of multiple devices. The I2C address for each individual device is derived from the port number specified in the command.

PLUGIN_WRITE was designed to extend any type of command to the system. The ESP Easy framework is totally unaware on the command syntax. For that reason, the plugin is solely responsible for any action on a command that it want to process.

A plugin is not limited as you state. But you need to code some stuff yourself when you need task related data within your plugin. The framework does not hide, block or do anything that prevents a plugin from accessing any data within ESP Easy.

Asume the following fictious command syntax for a plugin:

Demo,<taskname>

Then you would need something like this to make things work when a task is used to configure device specific settings:

    case PLUGIN_WRITE:
      {
        String command = parseString(string, 1);
        if (command == F("demo"))
        {
          String taskName = parseString(string, 2);
          int8_t taskIndex = getTaskIndexByName(taskName);
          if (taskIndex != -1)
          {
            success = true;
	    // from here you can use the taskIndex to acces all stuff from the task that is referenced!
            // show something on serial to prove that this works:
            Serial.println(ExtraTaskSettings.TaskDeviceName);
            Serial.println(Settings.TaskDevicePluginConfig[taskIndex];
          }
        }
        break;
      }

So i think it's just a few lines of extra code....

ShardanX, 2017-10-08:
Agree...
I'm using a remote control circuit with two PCF8574, one as input for keys, one as output for leds.
Both connected to an ESP-01 via I²C works fine.

papperone, 2017-10-09:
Sorry I didn't mean to be bad :( I apologize if that was the perception!
I shoudl have dig down more into ESPEasy core code to understand it better and maybe reach the solution you pointed by myself...
Indeed I understand your point and yes that few lines of codes will make all that I thougt a limit indeed possible :) I will play implementing this in the NeoPixel plugin as I do have needs to drive different sets of LEDs connected to separate GPIOs.
Thanks again for your understading and crystal clear explanation on how "things works" in the ESPEasy framework!!

psy0rz, 2017-10-12:
No problem papperone, this can be confusing for newcomers indeed. Also the wiki was not clear about this at all. (plugin_write was in the wrong section and it suggested it was impossible to do task related stuff) 

I changed the ordering of the page and added the example mentioned by Martinus: https://www.letscontrolit.com/wiki/index.php/ESPEasyDevelopment

closed 2017-10-12 by psy0rz