letscontrolit/ESPEasy #517: Tables <tr> and <td> are not closed properly.

Wandmalfarbe, 2017-09-27:
Nearly all of the HTML tables on the web interface are horribly broken because the tags `<tr>` and `<td>` are not properly closed.

### Expected behavior

The table on the main page after the login should look like this:

``` html
<table>
    <tbody>
        <tr>
            <th>System Info</th>
            <th>Value</th>
            <th></th>
            <th>System Info</th>
            <th>Value</th>
            <th></th>
        </tr>
        <tr>
            <td>Unit:</td>
            <td>0</td>
            <td></td>
            <td>GIT version:</td>
            <td>v2.0.0-dev12</td>
        </tr>
        <tr>
            <td>Local Time:</td>
            <td>NTP disabled</td>
            <td></td>
            <td>Uptime:</td>
            <td>0 days 4 hours 30 minutes</td>
        </tr>
        <tr>
            <td>Load:</td>
            <td>19% (LC=24784)</td>
            <td></td>
            <td>Free Mem:</td>
            <td>20736 (7256 - sendWebPageChunkedData)</td>
        </tr>
        <tr>
            <td>IP:</td>
            <td>192.168.178.47</td>
            <td></td>
            <td>Wifi RSSI:</td>
            <td>-65 dB</td>
        </tr>
        <tr>
            <th>Node List:</th>
            <th>Name</th>
            <th>Build</th>
            <th>Type</th>
            <th>IP</th>
            <th>Age</th>
        </tr>
        <tr>
            <td></td>
            <td></td>
        </tr>
    </tbody>
</table>
```

### Actual behavior

Instead the table looks like this:

``` html
<TR>
  <TH>System Info
    <TH>Value
      <TH>
        <TH>System Info
          <TH>Value
            <TH>
              <TR>
                <TD>Unit:
                  <TD>0
                    <TD>
                      <TD>GIT version:
                        <TD>v2.0.0-dev12
                          <TR>
                            <TD>Local Time:
                              <TD>NTP disabled
                                <TD>
                                  <TD>Uptime:
                                    <TD>0 days 4 hours 30 minutes
                                      <TR>
                                        <TD>Load:
                                          <TD>19% (LC=24784)
                                            <TD>
                                              <TD>Free Mem:
                                                <TD>20736 (7256 - sendWebPageChunkedData)
                                                  <TR>
                                                    <TD>IP:
                                                      <TD>192.168.178.47
                                                        <TD>
                                                          <TD>Wifi RSSI:
                                                            <TD>-65 dB
                                                              <TR>
                                                                <TH>Node List:
                                                                  <TH>Name
                                                                    <TH>Build
                                                                      <TH>Type
                                                                        <TH>IP
                                                                          <TH>Age
                                                                            <TR>
                                                                              <TD>
                                                                                <TD>
```

### System configuration

ESPEasy Version: v2.0.0-dev12

TD-er, 2017-09-27:
Indeed, it looks strange.
However, when using an [online validator](https://validator.w3.org/check), it is not complaining about missing closing tags.
It only gives warnings about table rows with less elements than the first line.
It looks like these closing tags are optional.

Still, I think this should be fixed, but then for another reason.
Currently the webserver is creating enormously big strings, which are terrible for such a small memory footprint.
So I think fixing that issue should also make it easier to append closing tags.

Wandmalfarbe, 2017-09-27:
Now that you mention it, I found out that the [end tags are optional](https://www.w3.org/TR/html5/tabular-data.html#the-tr-element). Maybe they were left out because of space constraints. I'll leave it to the owners to close this issue.

psy0rz, 2017-09-27:
indeed optional and done on purpose.

closed 2017-09-27 by psy0rz