letscontrolit/ESPEasy #518: DEV12 and upload file feature

kuipers250, 2017-09-28:
Maybe this is not the place to ask, but I like to know where I can find new features of releases, like now the DEV12.
I read something about a file upload feature. But how can I use this?

Grovkillen, 2017-09-28:
https://www.letscontrolit.com/forum/

Grovkillen, 2017-09-29:
@psy0rz we need to close this (I could help you cleaning stuff like this if I got admin rights).

psy0rz, 2017-10-01:
@Grovkillen I would appreciate that. :)

I dont have permission to add you myself. @mvdbro could you add @Grovkillen to the Organisation and add him to the Collaborators team?

Please dont do merges or pushes (yet). Cleaning up issues like you said is ok off course. Adding labels and milestones to issues is ok as well. :)

Grovkillen, 2017-10-01:
Sounds like a good plan. :)

Grovkillen, 2017-10-01:
Take this to the forum. I close this now.

closed 2017-10-01 by Grovkillen