letscontrolit/ESPEasy #522: HT16K33 brightness via MQTT

nonsintetic, 2017-10-01:
Hello, I got the new HT16K33 plugin to work with an Adafruit 0.56'' 7 Segment LED Backpack and it's amazing! 
It would be great if we could set the brightness in settings, and even better if we could set it via MQTT. The library already has the function setBrightness so it would be a very quick add for someone familiar with the plugin system. 

PS:
If anyone googles this: the settings for the Adafruit 7 segment backpacks are:
- 1st / 2nd GPIO: just leave empty
- Address: 0x70
- Clock type: 7Seg HH:MM
- the fields 'Seg for XX:XX' should be (in order): 0 , 1 , 3 , 4
- Seg. for Colon: 2, value: 2 
- also remember to turn on NTP in advanced settings and set a server, that's where the time displayed comes from

nonsintetic, 2017-10-01:
I couldn't help myself so I took a few hours and added it myself. It works from console and MQTT. Here is the pull request:

https://github.com/letscontrolit/ESPEasy/pull/523

psy0rz, 2017-10-01:
well thats awesome, thanks :)

psy0rz, 2017-10-01:
since you did it in a way that cant break anything existing, its safe to merge, so i merged it. :)

closed 2017-10-01 by psy0rz

rhopkins7428, 2024-03-29:
> Hello, I got the new HT16K33 plugin to work with an Adafruit 0.56'' 7 Segment LED Backpack and it's amazing! It would be great if we could **set the brightness** in settings, and even better if we [...]

Thanks for this post, it helped me get my display running in 2024 :)