letscontrolit/ESPEasy #524: Explain "Integration time" in the documentation.

Wandmalfarbe, 2017-10-03:
### Steps to reproduce

When visiting the wiki/documentation for the TSL2561, there is no explanation of the setting *Intergation time*.

### Expected behavior

The setting should be briefly described in the documentation. I actually found a nice description from sparkfun that could be an inspiration for the documentation text: <https://learn.sparkfun.com/tutorials/tsl2561-luminosity-sensor-hookup-guide/using-the-arduino-library>

### System configuration

Git version: v2.0.0-dev12

ShardanX, 2017-10-03:
Hello,

The wiki is undergoing many changes at this time as it is changed from the old R14x-Versions to ESPEasy 2.0.0.  I'm writing on the TSL and some other wiki chapters atm to bring them up to date.

ShardanX, 2017-10-03:
I've uploaded the reworked wiki description.
Be aware that this is a bit of a "quick shot" and may need some fine tuning and formatting.

Grovkillen, 2017-10-04:
I think this is good as of now. Great work @ShardanX

closed 2017-10-04 by Grovkillen