letscontrolit/ESPEasy #526: Feature Request - Get variable or device value from Domoticz

MiloshCZ, 2017-10-04:
Hi, it would be nice to add action for get variable or device value from Domoticz. For example: for setup output after startup.

Thanks

TD-er, 2017-10-04:
Like to automatically setup the configuration of the plugin after setup of the Domoticz controller?

Does Domoticz support something like that already?

MiloshCZ, 2017-10-04:
Domoticz doesnt support it, but you can request actual value back from domoticz.
Maybe new command for usage in rules, something like:
`DomoticzGet <idx>,<ValueName>` where "idx" will be device id, "ValueName" name of value (for example Temp, BatteryLevel, Humidity etc.)
`DomoticzVarGet <idx>` where "idx" will be User variable id
https://www.domoticz.com/wiki/Domoticz_API/JSON_URL's#List_one_variable

adrianmihalko, 2018-03-01:
This would be great. I am using EspEasy to control dimmable LED lights it would be great to set startup  state from Domoticz (after power outage etc).

tonhuisman, 2022-05-07:
This sounds like the recipe for P029, Domoticz helper, and as that is already supported, this can probably be closed.

closed 2022-05-07 by TD-er