letscontrolit/ESPEasy #527: ESPEasy is not handling MQTT very well

ArJay60, 2017-10-06:
It seems that the later versions of Domoticz have a change in the MQTT message. The item Description is added and ESPEasy is not handling this well.
I am using ESPeasy build 142 using Domoticz MQTT as protocol

MQTT message with older version of Domoticz:
{
"Battery" : 255,
"RSSI" : 12,
"dtype" : "Light/Switch",
"id" : "0001408A",
"idx" : 58,
"name" : "HK Schemer Acht",
"nvalue" : 1,
"stype" : "Switch",
"svalue1" : "0",
"switchType" : "On/Off",
"unit" : 1
}
MQTT message with newer version of Domoticz:
{
"Battery" : 255,
"RSSI" : 12,
"description" : "", <------ removing this line from the message to ESPeasy solves the problem.
"dtype" : "Light/Switch",
"id" : "0001408A",
"idx" : 58,
"name" : "HK Schemer Acht",
"nvalue" : 1,
"stype" : "Switch",
"svalue1" : "0",
"switchType" : "On/Off",
"unit" : 1
}

Can someone of the developers make MQTT within ESPeasy less sensitive for these changes in message layout on MQTT from Domoticz.

Grovkillen, 2017-10-06:
Dublicate: #513

closed 2017-10-06 by Grovkillen

mvdbro, 2017-10-08:
Check if you're using our patched pubsubclient

ArJay60, 2017-10-08:
Because I used precompiled versions of EasyESP i don't know which version of pubsubclient is within the 142 build. I have grabbed the latest version of EasyESP and that one doesn't have the problem. Probably because of revised pubsubclient ??
My problem is solved.