letscontrolit/ESPEasy #528: Flash tool change

ShardanX, 2017-10-07:
More and more devices are comming up that need to be flashed  with DOUT mode.
After Sonoff basic and S20 now some ESP-01 show up that have to be flashed with DOUT.

As other devices including older ones still use quad speed flashing i suggest to
implement a button or click box for setting the accurate flash mode.

Grovkillen, 2017-10-08:
I put it for the 2.1.0 version since I think no developer has the time to add this until after the 2.0.0 release. This is a top priority issue I think.

jahnroux, 2017-12-09:
Maybe as a temporary solution add documentation on how to set dout mode with esptool.exe (if functionality exists). I tried to add the command -fm dout like with esptool.py but it did not work. Needed to use the expressif tool to flash espeasy 2.0 into the latest batch of sonoff basic.

Grovkillen, 2017-12-09:
You need to flash them with the ESP8285 bin file. They seems to be a
different kind than the older sonoff which liked the ESP8266 bin file. That
info is added on the wiki at least.

Grovkillen, 2017-12-09:
```-bm <qio/qout/dio/dout>``` | Set the flash chip interface mode. Default is QIO. This parameter is stored in the binary image header, along with the flash size and flash frequency. The ROM bootloader in the ESP8266 uses the value of these parameters in order to know how to talk to the flash chip.

jahnroux, 2017-12-10:
Correct, esptool.exe by default uses the "keep" flash mode which uses whatever your bin is compiled with. Thanks for the help.

Grovkillen, 2017-12-11:
I will try to get time to finish my new interface for the flash tool. This will let you override the bin settings and also config the devices over serial making the AP step a thing for the beginners.

jahnroux, 2017-12-12:
What language is the interface written in? Maybe I can help code.

Grovkillen, 2017-12-12:
It's a script language called Macro Scheduler. I have used it for years to automate stuff in Windows.

uzi18, 2017-12-12:
@Grovkillen : Did You tested AutoIt?

Grovkillen, 2017-12-12:
Yes, not as complete as MS in my opinion.

uzi18, 2017-12-12:
@Grovkillen  but AutoIt is for free and MS is realy expensive ...

Do You want a tool to flash properly bin image or something more?
Maybe just provide an esptool wrapper for known hardware to force dout on related modules.

jahnroux, 2017-12-12:
I started writing a wrapper for esptool in c# for myself. but would have .net framework dependencies. If that't not an issue I can provide a simple tool for flashing, but with an advanced options to enable dout etc.

Grovkillen, 2017-12-12:
@uzi18 I wouldn't say €90/year expensive, sure it's not free but I love how the developers are constantly keeping it up to date.

To both: yes it will be a wrapper for the esptool.exe + communication over serial for post-flash configuration.

All means of help is appreciated and if you are able to fix a better wrapper faster then go for it :)

psy0rz, 2017-12-18:
i dont prefetr stuff with extra dependencies myself, but its better than having nothing at all :)

uzi18, 2017-12-18:
esptool-ck isn't wrapper already?
Does it support all we need?

TD-er, 2018-03-04:
I just set the `board_flash_mode = dout` in PlatformIO for the 1M builds.
Then the sketch itself will use the dout mode to access the flash.

Grovkillen, 2018-03-25:
Should we close this one?