letscontrolit/ESPEasy #529: Simple fix for MH-Z19 B version.

TD-er, 2017-10-07:
I added pull request #525 a few days ago and it is marked for version 2.1, which is fine.
But there is a only a very small fix needed to get the "B" version of the sensor working in the 2.0 branch.
So is it still possible to add this small fix and then later merge my pul request for the 2.1 release cycle?

The small fix is to allow samples with a "S" value of 0 or 64, instead of only 64.
Then the users with the "B" version can also use the 2.0 builds without having to code and compile themselves.

It is just a small change in line 295:https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P049_MHZ19.ino#L295

psy0rz, 2017-10-09:
ah thats fine indeed. thanks.

psy0rz, 2017-10-09:
thanks, i changed it in 0dfa98cacc71e68d0c8aea6377e1e428875d064f

closed 2017-10-09 by psy0rz

psy0rz, 2017-10-15:
dont know what happend...cant find the commit nor the change :S

psy0rz, 2017-10-15:
well now its changed for real :)