letscontrolit/ESPEasy #533: Add support of Blynk server.

ReliableBob, 2017-10-09:
Hey.
Really need the support of Blynk server. I know that such development was carried out and the plug-in was tested, but for some reason this plug-in is not in the list of supported controllers. 
Can we hope that in new versions Blynk support will be added?

To1952P, 2017-10-10:
Agree with Reliable Rob. I believe Blynk is the most sophisticated app available

psy0rz, 2017-10-12:
related to #92 

I will look into it or provide a solution with the http advanced controller.

@krikk is that possible?

krikk, 2017-10-13:
had a quick look into the [Blynk protocol](https://github.com/blynkkk/blynk-server/blob/master/docs/README_FOR_APP_DEVS.md#protocol-messages) and by default it is a binary protocol, so not really do able with http advanced, but there is also a [http api](http://docs.blynkapi.apiary.io/#) so this should be possible with http advanced but i am not sure if this is full featured... 

so blynk controller which implements the binary protocol seems to be a valid request... :)

psy0rz, 2017-10-15:
ok thanks :)

fverschure, 2018-01-30:
any progress? Would love to be able to use ESPEasy with Blynk for an IoT workshop. This would eliminate the need use an IDE.

ReliableBob, 2018-01-30:
Hey. I'm sorry, it's been a long time and now I'm not interested in Blynk. I am now with the iobroker and Espeasy)))

fverschure, 2018-01-30:
Hi Riliablebob, thanks for your response. So does this mean that development for this module is stopped?
I read in another thread that it was incorporated in the testing plugin set? So can i use it with one of the builds and how?

ReliableBob, 2018-01-30:
I can not answer this question because I'm not a developer

psy0rz, 2018-02-02:
We're just too busy with fixing v2.0 stuff. Maybe you could try to get Blynk to work with the http advanced module yourself? Also vote up the first comment by using the thumbs up. This helps us to see how many people are waiting for this feature. (or for documentation to use it)

fverschure, 2018-02-02:
hi, thanks for your response. I'll take a look into the http advanced module and the blynk api.

fverschure, 2018-02-09:
Hi, I looked into the http advanced module. I published my findings in the thread referred to in this thread: 
[](https://github.com/letscontrolit/ESPEasy/pull/92)

psy0rz, 2018-02-12:
w00p..@TD-er decided to port the blink plugin, so it will be in mega. so wait for the next nightly mega-release or compile it yourself. :)

closed 2018-02-12 by psy0rz