letscontrolit/ESPEasy #539: Generic RS232 plugin

Grovkillen, 2017-10-13:
My projector, my receiver, and may tv all have serial communication over RS232. I have an idea of a generic RS232/Serial plugin which let you control devices that have this interface, and maybe even monitor its status. Would it be possible to create such a plugin or does it need to be unit specific? My hope was to have a ESP Easy unit connected to these devices and have them controlled using MQTT and rules? Is this to much to ask for? :)

Just as info:
My tv got [this connection](https://cnet1.cbsistatic.com/img/5CC1SMjuvzxyB5ra-BXcIaRHamE=/830x467/2010/08/03/f8547916-67c3-11e3-a665-14feb5ca9861/34107674-2-1333-BK-3.jpg) ([EX-link aka Service Port](https://wiki.samygo.tv/images/4/49/Img.jpg)) and accept [these commands](https://support.justaddpower.com/kb/article/16-samsung-rs232-control-exlink). 
My projector got [this connection](http://ksassets.timeincuk.net/wp/uploads/sites/54/2007/03/4343-IMG1778s-1.jpg) and accept [these commands](https://files.support.epson.com/Epson_Handbook/html/p85_rs232.html). 
My receiver got [this connection](http://hometheaterhifi.com/volume_9_3/images/denon-5803-receiver-rear-panel-large.jpg) and accept [these commands](https://github.com/phillipsnick/denon-avr).

[Inspiration video :)](https://www.youtube.com/watch?v=bZS8FMjxmts)

The cables are easy to make yourself but the hard part might be to create a generic plugin? Feel free to comment! :)

TD-er, 2017-10-14:
What are the signal levels required for these devices?
RS232 is so old, its original standard even required negative voltage signals, but later devices were more 'friendly' and accepted TTL levels.

But still, TTL levels are up-to 5V, so you maybe also need a level converter.
And you may need a lot of settings, so maybe some kind of profile per device may be useful.

Grovkillen, 2017-10-14:
Ah, yes different profiles would be a way of doing it. I haven't studied my particular devices yet so I don't know the voltage levels.

mikeluyten, 2017-10-16:
I'm currenty doing this via http commands
http://10.0.0.15/control?cmd=serialsend,* 0 IR 001

This sends the command "* 0 IR 001" over the tx pin;
Which is then hooked up to a TTL to RS232 converter from AliExpress
(https://www.aliexpress.com/item/MAX3232-RS232-to-TTL-Serial-Port-Converter-Module-DB9-Connector-MAX232/32353317234.html)

Regardless of the connector, if it says serial or RS232, you will need a TTL to RS232 converter of some kind due to the floating ground used in RS232 signals, and the potentially -25v to +25v.

Grovkillen, 2017-11-21:
I have now installed this at home but I'm unable to control my projector. I suspect that the baud rate might be an issue? The projector expects 9600 but my TTL-RS232 is okay with higher baud rates than that. What's the baud rate ESP Easy is communication on by default?

mikeluyten, 2017-11-21:
11200 I believe is the default, you can change this under Tools -> Advanced -> Baud Rate.
Then restart your esp

TD-er, 2017-11-21:
Make sure you connect the Rx of the ESP board to the TX  of the projector and vice verse.

And for RS232 communication, you also should take the number of start and stop bits and parity into account.

Sometimes you have to add some dummy value to "wake" the device and allow for synchronization.
Can you control the projector from a PC? If so, you can try to capture the data using some tools like GPSgate, which split the COM port in software.

Grovkillen, 2017-11-21:
Thanks for your suggestions. I will try to control it from a PC first. Good advice. It might be something that I need to do to "trigger" the activity on the RS232 side on the projector.

TD-er, 2017-11-21:
A very commonly used value to 'pre-trigger' communications is to send several 0xFF (8 times '1') or 0xA5 (10100101).
So if you see that at the start of a message, you know why that's needed :)

rr10, 2018-08-22:
Summarize of the problem/feature request
would like to make an adapter from rs232 to mqtt.
I need this adapter to read the data provided by a barcode scanner with rs232 port (female). Can I do it with espeasy? I also bought a rs232 (female) to ttl (max232) converter and a rs232 male to male adapter.
how can I do?

https://www.aliexpress.com/store/product/MAX3232-RS232-to-TTL-Serial-Port-Converter-Module-DB9-Connector-MAX232/1504763_32842352700.html

https://www.aliexpress.com/store/product/9-Pin-RS-232-DB9-Male-to-Male-Serial-Cable-Gender-Changer-Coupler-Adapter/502834_686740151.html

http://www.letscontrolit.com/forum/viewtopic.php?f=2&t=5697

do not have experience of plugins of this type, I would like to learn a lot but at this moment I have short time. if someone wanted to implement the plugin I would be willing to make an economic contribution and I would like the plugin to be used by the whole community

ShardanX, 2018-12-06:
Agree.
A two-way ser2net would be very appreciated so you could send and receive RS232 from ant to WiFi,
It's not only TV sets, HiFi and such, we have some sensors using serial for example. Some devices are just rare so a plugin for every device would be overkill in my opinion.
Most home controls like FHEM, OpenHAB domotics etc can be extended easily with some lines of code.
So it should be possible to just send/receive raw data via the ESP and let the controler do the work.
MQTT would be a way, maybe http too.

BTW it is same with I²C. A generic I²C would allow to test and use sensors etc that are not implemented.

Both would be a powerfull extension that makes ESPEasy really universal

TD-er, 2018-12-06:
Would it also make sense to make the ESP as "wireless bridge"?
One on the "PC" side, and one on the device side.

ShardanX, 2018-12-08:
Just some ideas i just ran into: (for a leter release, really)

On long view ressources are limited.. Flash isn't unlimited, and ram is a even more limited ressource. They can't take all and every possible device users may want on a long time view.

One way out might be a compilation on demand, just compiling the plugins needed into a binary. Complex to realize via a web page, and definitely not the darling for most users, on top self compiling this way is nearly impossible to support.

Such generic plugins might be a way out. They should be reusable so roughly one generic plugin for I²C could serve nearly any I²C device. There should be configuration samples so the plugin is easy to handle. Optimal case would be loadable configurations similar to loading rules from a file...

One could even think of a combination, use the nowadays plugins for the most wanted devices (BME-Sensors, I²C-Displays...) and do the other stuff with generic plugins. This would allow a smooth change from today's state to generic plugins.

(To keep that clear - it's just a raw idea to be discussed....)

Have a nice 2nd Advent, all of you.

uzi18, 2018-12-08:
Just teach controller how to send/receive serial data, for mqtt it could be mqtt import plugin

tparvais, 2019-06-04:
Hello
How can I send a serie of Hexa values to control my videoprojector ? 
I need to send 0x08 0x22 0x00 0x00 0x00 0x16 via serialsend command ?
thanks

Tobias007, 2019-06-17:
Hi, you can pass it via URL:
http://192.168.4.1/control?cmd=serialsend,%A0%01%A1
however it doesn't work, if you want to send 0x00 as this is interpreted as end of string...
I fear without developing e.g. a serialsendhex command it's not possible to send 0x00...

foice, 2020-01-25:
> My projector got [this connection](http://ksassets.timeincuk.net/wp/uploads/sites/54/2007/03/4343-IMG1778s-1.jpg) and accept [these commands](https://files.support.epson.com/Epson_Handbook/html/p85_rs232.html).
 
You have got a broken link for the commands accepted by your projector (which seems to be a EMP TW-700, isn't it?)

Grovkillen, 2020-01-25:
Yep it's a TW-700. I haven't gotten around to fiddle more with this though.

foice, 2020-01-25:
Got you! : )
Are you saying that you were not successful in operating it via RS232? I have just connected it to my FTDI232 but the projector seems to not really be talking to the chip. I can set the speed at 9600 but so far I get nothing when I turn on the projector on my terminal. Any suggestion on how to connect the projector to my OS X or linux terminal?

tonhuisman, 2023-09-02:
With all the serial communication stuff we have available, this is no longer relevant. Can be closed.

closed 2023-09-02 by TD-er