letscontrolit/ESPEasy #554: When node crashes it is always at x:59 hour

TD-er, 2017-10-18:
I have several nodes running with ESPeasy and one of them (connected to my laptop's USB) is sometimes crashing. There is a display connected to it and it uses a syslog server.
When the node crashes, it is frozen and thus the display is not updated and showing for example 04:59. Also no updates on the syslog server are received after that moment.

So I was thinking what could cause such a crash.
Since others are working fine for weeks, it might be related to that board and/or its power supply.
But that is also my testing board, so it might also be my own code, or something related to the devices which are not connected to other nodes.

Is there something running in ESPeasy at or around the start of an hour, which could take just enough resources the power converters (on that board) cannot deal with? Or perhaps triggers some bug?

Grovkillen, 2017-10-18:
Good finding.

TD-er, 2017-10-18:
I noticed the crashing node is having a lot less free memory. 
For example: 

> Free Mem: | 15936 (4896 - sendWebPageChunkedData)

So I am testing something to reduce the amount of memory used by the webserver.
The webserver is appending to a string, which may lead to heap fragmentation.

My tests can be seen here: https://github.com/TD-er/ESPEasy/tree/feature/reduce_memory
Just to get an idea of the amount of data used in the webserver, I added a print at the bottom of each page.

Grovkillen, 2017-10-23:
I've added this to the 2.1.0 milestone since we really need to get the 2.0.0 out into the wild.

TD-er, 2017-10-23:
I don't expect to find what's causing this within a few days, so that's perfectly fine.
I hope someone else experiences a similar issue, just to narrow down where to look.

TD-er, 2017-10-27:
Just an update, I moved this sensor to another (external) power supply, just to exclude the option the laptop is doing something with the USB power.
The reboot/crash still happens (always?) at around 6 am. (+/- 2 minutes)
I was probably confused in thinking it also happens at other hours, because of the displayed time on the display, but that has a timezone offset.

TD-er, 2017-10-29:
This morning, it also restarted at around 6 am, so given it was the change to winter time, it may be something external causing this (something at the router???)
My ESP device does not know about summer/winter time, so it is rather strange it was happening at the 'new' 5:59 am.

TD-er, 2017-10-31:
OK, this bug is officially a [PEBKAC](https://www.urbandictionary.com/define.php?term=pebkac).

For everyone's amusement, here a cut'n-paste of the cronjob of one of my machines:
`0 6 * * * /usr/bin/curl -m 30 --silent "http://192.168.1.75/?cmd=reboot" > /dev/null 2>&1
`

/me is now going to fold a donkey hat and moving to the corner of the room...

One positive thing... the reboot command works just fine :)

closed 2017-10-31 by TD-er

Grovkillen, 2017-10-31:
:P

uzi18, 2017-10-31:
Nice =)

psy0rz, 2017-11-14:
thanks for testing the reboot command :P