letscontrolit/ESPEasy #555: Support for different Domoticz webroot URL's

Elmardus, 2017-10-20:
### Steps to reproduce
How can we trigger this problem?
I have set my domoticz to have a different webroot, so it is running at
www.exampleserver.com:80/domoticz
instead of www.exampleserver.com:8080

ESP easy gives an error when trying to connect, because of the missing support for a different webroot.

### Expected behavior
Tell us what should happen?
An extra configuration option is needed to set a custom controller URL 

### System configuration

Hardware:
ESP8266-12E

Grovkillen, 2019-10-11:
I will close this, open if its still a valid issue.

CV8R, 2020-01-29:
Hi, this issue affects my setup where I am using a different webroot path in Domoticz.  Same symptoms as original report.  If I run Domoticz from command line without -webroot then everything works as expected, if i run from command line with -webroot then ESP easy gives an error when trying to connect.

TD-er, 2020-01-29:
Can you give the error?
I assume it is with the Domoticz HTTP controller and not the MQTT?

CV8R, 2020-01-29:
Correct - Domoticz HTTP Protocol using IP Address as the connection method.

However, on testing more thoroughly, this is interesting.

When running Domoticz without webroot grepping for messages from the ESP Easy on 14.78.5.91.  Domoticz logs the incoming connection.

`domoticz -www 8080 | grep 14.78.5.91`

Result:
`2020-01-29 20:36:59.274  Status: Incoming connection from: 14.78.5.91`

When changing the username for the Domoticz server in ESP Easy to an incorrect one, Domoticz logs the failed auth:

`2020-01-29 20:33:26.556  Error: [web:8080] Failed authentication attempt, ignoring client request (remote address: 14.78.5.91)`

There are no connection messages in the ESP Easy Web Debug @ Debug dev level failed or otherwise.

When stopping the Domoticz server I see the connection failures in the ESP Easy Web Debug

46500: EVENT: Clock#Time=Wed,19:42
50613: SW   : State 0.00
50616: EVENT: Relay#State=0.00
50657: HTTP : C001 connection failed (23/0)
50782: HTTP : C001 connection failed (24/0)
50907: HTTP : C001 connection failed (25/0)
51030: HTTP : C001 connection failed (26/0)
51157: HTTP : C001 connection failed (27/0)
51277: HTTP : C001 connection failed (28/0)
51403: HTTP : C001 connection failed (29/0)
51531: HTTP : C001 connection failed (30/0)
51676: HTTP : C001 connection failed (31/0)
51801: HTTP : C001 connection failed (32/0)
51921: HTTP : C001 connection failed (33/0)
60613: SW   : State 0.00
60616: EVENT: Relay#State=0.00
62055: WD   : Uptime 1 ConnectFailures 32 FreeMem 21488 WiFiStatus 3

When running Domoticz with a modified webroot with:

`domoticz -www 8080 -webroot homeautomation`

There are no connection messages in the Domoticz log, like no communication makes it to the Domoticz server at all.

However, the interesting part is that there are no HTTP : C001 connection failed messages in the ESP Easy debug log.

I have tried Use Hostname by adding ServerIP/homeautomation/ hostname/homeautomation/ and when doing this I see connection failed messages.

It appears there is no way to append the webroot url to the controller address in the ESP Easy GUI.  Also, the connection failure messages are misleading when the server is up, but the connection does not have the correct path.

Cheers
CV8R

TD-er, 2020-01-30:
> It appears there is no way to append the webroot url to the controller address in the ESP Easy GUI. Also, the connection failure messages are misleading when the server is up, but the connection does not have the correct path.

That's correct, I noticed something similar.
The connection failures are dealt with in a different way per controller.
I have tried to fix this in [this commit](https://github.com/letscontrolit/ESPEasy/commit/91b7b3b9a2c1de78073dc046983ca3877f3c3cb5) but for some reason (unknown to me), this commit triggers crashes on some ESP modules.
In a test build I am trying here at home, I have tried digesting it for a while and have not found what makes it crash. (and only on some modules)