letscontrolit/ESPEasy #557: Task reduction for PCF8574/MCP2308/MCP23017

ShardanX, 2017-10-21:
It's well known that a port expander used as input takes up a lot of tasks. It is even not possible to use all 16 ports of the MCP23017 as 12 tasks != 16 ports.

I suggest an enhancement to make this easier to handle taking up just one task

First solution: Add a "one-task-mode" that gives back the pin number of the pressed key.
So if the key on pin P3 is pressed give back "3".
This solution has one disadvantage: It gets "stupid" if if more then one key is pressed accidentally or
intentionally. The plugin will have to handle this in some way, for example sending two values one by one.

Second solution: Add a "one-task-mode" that just sends the byte or word from the GPIO expander.
For exampe if the key at P3 pin is pressed it should send a "00001000" (16) value.
A small plugin in domoticz or FHEM can do the resolving easily by ANDing 
("If %value% AND 16 then......").
This way you may press several keys at once without driving the plugin crazy.

With this way and some effort in domoticz or FHEM even big remote controls are possible
for example. For 4 MCP23017 you need 4 tasks giving 64 keys......

papperone, 2017-10-22:
well I can't be more than agree...
I am designing an 8 relays and 16 relays control board using exactly MCP23017 but I'd like to have as well 8 and 16 input pins, and current ESPEasy will make those hardware useless :(
Soon I'll send PCB for manufacturing and I could have a nice "test hardware" for such plugin update :)

vikasjee, 2017-11-30:
I am designing a 16 Buttons + 8/12/16 Relays/LEDs control board using MCP23017. This feature is urgently needed before i get my hardware done. Any Release date for 2.1.0 as yet?

TD-er, 2017-11-30:
This should be implemented in the specific plugin for this chip?
It sounds like it should be implemented not on plugin level, but on a higher level. But on the other hand I don't know of any other similar chips that would benefit from some (C++ like) switch statement logic like this.

ShardanX, 2017-11-30:
I don't know any other plugin either, i think it should be done on plugin level.
This shouldn't be too difficult. Afaik the plugin reads the register from the chip and "cuts out" the GPIO pin wanted by the task. An additional code to re-assmble the output into one byte/word is a bit of nonsense in my eyes - the byte/word value for all pins together should already exist somewhere in the plugin. It needs a choice inside the task if the user wants to read a single pin or the whole byte/word value.

TD-er, 2017-11-30:
I am not (yet) familiar with the plugin.
But if I understand you correctly, adding just an extra "ouput" of the total 8 or 16 bits in one value is also a viable fix?
Are bitwise operations allowed in tasks? (to select a single bit)

ShardanX, 2017-11-30:
I'm no programmer at all but as far as i can see from the chips datasheets is that on reading the chip should give back the whole internal register via I²C. The task includes a port number. The chip's I²C address and port is evaluated from this number. If you chose port 10 from a PCF8574 (8-port) the second port of the second chip address is the value shown in the task. If the plugin works this way it should be possible to give back the whole register value easily.  As said i'm no coder, i might be wrong.

ShardanX, 2017-11-30:
Tried to understand a bit of the source.

_If i'm right the input is read and broken down in this part:
  // get the current pin status
  Wire.requestFrom(address, (uint8_t)0x1);
  if (Wire.available())
  {
    state = ((Wire.read() & _BV(port - 1)) >> (port - 1));
  }
  return state;
}_

I assume that the line
 _state = ((Wire.read() & _BV(port - 1)) >> (port - 1));_
filters the addressed port out of the byte read from the PCF8574 register.

TD-er, 2017-12-01:
This _BV macro was also unknown to me, but it [appeared](http://www.atmel.com/webdoc/avrlibcreferencemanual/FAQ_1faq_use_bv.html) to be a simple macro to get the byte value based on a simple bitshift.

So Wire.read() will return a byte (8 bits) to which a mask is applied of one single bit.
Then it is shifted back.
So in normal English, the state will contain either 0 or 1, based on looking at a single bit in the returned value.

More readable would be:
```
if (!Wire.available()) {
  return -1;
}
if ((Wire.read() & _BV(port - 1) == 0) {
  return 0;
}
return 1;
```
The rest of the code in the read function is decoding the I2C address and portnumber of the chip from a single byte.
In short, it should not be hard to read all bits at once, but it may need more changes in the code than one would expect at first sight, since the plugin design (interface with the rules) may need to be changed.

vikasjee, 2018-06-18:
When is this expected to be released?

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er