letscontrolit/ESPEasy #558: WPA KRACK vulnerability

TD-er, 2017-10-21:
According to [this forum post](https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3673), it should be quite easy to update to apply a patch for the WPA KRACK vulnerability.
See also: https://github.com/esp8266/Arduino/issues/3725

Given that these boards should not be connected to the internet, it may be less of an issue??????

Is it too much of an issue to release it with 2.0.0?

I will also look into it, to see how much impact it may have and perhaps create a pull request.

beicnet, 2017-10-22:
I don't know, but no one will WPA KRACK your ESP device, it's almost like 2k compatibility issue with BIOS back time. Anyway, that means the whole core need to be upgraded to 2.4.0-rc build and it's full of bugs.

TD-er, 2017-10-22:
@beicnet I agree.
If indeed the core is still not mature enough, we should wait for the upgrade.

beicnet, 2017-10-22:
@TD-er It was only what I tought, but anyway it's not like military or commercial degree devices. Also my devies are not directly connected to the internet, they are all on the intranet. And I have a beautiful/useful video for you (worth any cent to watch) [Vlog #58: Wires NOT WiFi for home automation!](https://www.youtube.com/watch?v=1JDHWPm-ltY)

psy0rz, 2018-01-12:
Arduino core 2.4.0 is still in staging in platformio: https://community.platformio.org/t/esp8266-arduino-krack-fix-compile-using-platformio/2747/4

Also there where some memory issues we might need to fix if we're going to 2.4.0. 

So for now we stay on 2.3.0 until 2.4.0 is considered stable in platformio, and focus on getting ESPEasy v2.0.0 stable.

I want to fix our current bugs first, before introducing potentially new issues with 2.4.0. (which in itself is still considered staging)

PR #2559 (2019-08-26): [WiFi] Improve WiFi reconnect stability and fix initial setup
Simplified the WiFi connect state machine.
Fixed setup of initial WiFi configuration.

AP mode is now kept as long as client is connected.
AP mode will be started immediately if there are no valid WiFi credentials, or after 20 seconds of no wifi connection.
AP mode will be disabled 60 seconds after starting it, or after the last client disconnected from the AP.

Fixes #2561 
Fixes #2338 
Fixes #2203 
Fixes #1987 
Fixes #1737
Fixes #1640 
Fixes #1565
Fixes #1479 
Fixes #1098
Fixes #558
closed 2019-08-26 by TD-er