letscontrolit/ESPEasy #562: Multiple access points create confusion when using the Setup Wizard

Grovkillen, 2017-10-25:
If you run the Setup Wizard and you have a network with multiple access points they will all be listed like:

```
NetworkName
NetworkName
NetworkName
NetworkName
```

And you will just have to pick one in random. But if we could also have the RSSI signal strength and possibly the **B**SSID value presented it would be easier to identify the closest access point. The above list would look something like this:

```
NetworkName [24:a4:3c:9e:d2:84] -45 dBm
NetworkName [24:a4:3c:ae:df:83] -87 dBm
NetworkName [dc:9f:db:30:c1:7a] -65 dBm
NetworkName [00:19:3b:99:e2:80] -90 dBm
```

On another (closely related) note, today I use the RSSI system value in rules on a once-an-hour basis to check if the signal is below a certain value a wifi disconnect and wifi connect is performed. This could (should) be implemented int the source code.

The PR #101 was rejected it seems?

Grovkillen, 2018-03-25:
This could now be implemented since we now have the info as system variables.

TD-er, 2018-03-25:
With the current (recently added) WiFi code, the reconnect would first be based on the BSSID of the last connection.
This was to keep connections stable when using repeaters.
Also as soon as a disconnect occurs, a reconnect is initiated.

- First attempt is at the last known BSSID
- Next 2 attempts is using the credentials of the last connected accesspoint. 
- Next 3 attempts is to the 2nd defined network credentials, when available and valid.

Also the RSSI value only is an indication of the signal strength, not of network stability. Only an unintentional disconnect is an indication of that.
There is some correlation between RSSI and stability and RSSI and connection speed and last there is also some correlation between stability and connection speed.
But these correlations have no direct relation and I know how paradoxical that may sound.

Showing BSSID and RSSI at the first setup page can be done without much effort. The values are collected in a scan if I remember correctly.

Grovkillen, 2018-03-25:
Sounds good my friend!

Grovkillen, 2018-03-25:
The idea behind this request was to show the user that these listed network names are multiple by a reason and not mistakenly listed multiple times. Make sense? :stuck_out_tongue:

closed 2018-04-14 by Grovkillen