letscontrolit/ESPEasy #564: Feature request: Add "battery" parameter for Domoticz HTTP/MQTT controllers

choxnox, 2017-10-26:
Domoticz supports battery level indicator which is useful for battery-powered applications such as weather stations or sensors in general. To make battery level report work, all that needs to be done is passing "battery" parameter to Domoticz controller indicating battery level in %.

Unfortunately, ESPEasy v2.0 in it's current state generates Domoticz-friendly HTTP request in such way that parameters passed are hardcoded. This means there is no option to pass custom parameters to Domoticz whose values would be sourced from values got from connected hardware.

For example, I've several weather stations made using Wemos D1 mini Pro + Wemos SHT30 shield + Wemos battery shield which can very easily read battery voltage using A0 input. Maybe the easiest solution is to modify Analog plugin to add additional checkbox field simply stating whether this analog hardware is battery indicator or not, and then to take this hardware's value into consideration when generating Domoticz HTTP URLs.

Any other thoughts or ideas about this?

TD-er, 2017-10-26:
Does this parameter have to be in the same message as the measurements, or can it also be a separate "device" which sends its own messages?

Grovkillen, 2017-10-26:
Just for info: you can make your own custom messages using rules. It's documented in the wiki https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules#Custom_reports_to_Domoticz_with_own_IDX

But a battery status as a device might be a more generic (user friendly) approach to your suggestion.

choxnox, 2017-10-26:
@TD-er: Unfortunately, `battery` parameter must be sent altogether with sensors data. That means leaving out `svalue` parameter will return error from Domoticz, and if you populate that parameter with any dummy value, then it's going to ignore the real value of the sensor.

@Grovkillen: I've tried to make it work via Rules just like in that example (it generates correct URL) but the problem is Domoticz server is password protected and yet `SendToHTTP` doesn't support authentication.

TD-er, 2017-10-26:
Can you give an example of a proper formatted message?

And I guess it should be some global setting, or at least a special kind of device, which attaches its measurement value to each (or selected set of) MQTT/HTTP message or specific (message) controller.

Grovkillen, 2017-10-26:
@choxnox I guess using a MQTT broker would work though.

choxnox, 2017-10-26:
Here is a valid URL for SHT30 (temperature & humidity) reporting:

`http://{hostname}:{port}/json.htm?type=command&param=udevice&idx={deviceIdx}&svalue={temperature};{humidity};0&battery={batteryPercent}`

Global setting sounds great but I think it should also have an option to enable calibration and/or formula correction just like in the `Analog` plugin (to correct possible fluctuations in the analog reading). Take a look at this thread why global setting would be better: https://www.letscontrolit.com/forum/viewtopic.php?t=1621

As it can be seen on the photo later in that thread ([https://www.letscontrolit.com/forum/download/file.php?id=1054](https://www.letscontrolit.com/forum/download/file.php?id=1054)) having a global setting would open a door to the new feature such as displaying battery level percentage directly in the ESP's web interface regardless of the integrated controller.

choxnox, 2017-10-26:
@Grovkillen: Unfortunatelly, MQTT parameters are also hardcoded which means `battery` parameter is ignored.

Grovkillen, 2017-10-26:
@choxnox if you use the publish command in a rule it would work I suspect. (As described in the rules wiki example I linked to)

choxnox, 2017-10-26:
@Grovkillen: I know it should work but it doesn't. For example this is copy-paste from the log:

```
570706 : ACT : Publish domoticz/in, {"idx": 10, "svalue": "19.48;49.31;0", "battery": 88}
571199 : MQTT : Topic: domoticz/out
571199 : MQTT : Payload: { "Battery" : 255, "RSSI" : 12, "description" : "", "dtype" : "Temp + Humidity", "id" : "82010",
```

Log is truncated but you can see that `battery` parameter is ignored (Domoticz displays correct temperature and humidity values). I also went through the Domoticz MQTT controller code in `_C002.ino` and it's clear that parameters are hardcoded to only use 5 of them: `idx`, `nvalue`,`svalue`, `command` and `switchcmd`. Everything else gets ignored.

Grovkillen, 2017-10-26:
Aha, I understand. Thanks for explaining!

enesbcs, 2017-12-07:
As i see "Battery" and "RSSI" can be sent by Publish rule command... except, that they are Case-sensitive on the Domoticz side, and for some reason, the ESPeasy convert all sent parameters to lower case. :(

I tried to use the following ESPEasy rule command:
Publish domoticz/in,{"idx":55,"nvalue":0.00,"svalue":"Off", "RSSI": [RSSI#RSSI]}

On the domoticz side i see this result:
2017-12-07 21:28:54.526 MQTT: Topic: domoticz/in, Message: {"idx":55,"nvalue":0.00,"svalue":"off", "rssi": 5}

I guess that Domoticz does not accept "rssi" with lower case...
When i send the same MQTT string with mosquitto_pub as RSSI uppercase, than Domoticz handle this correctly.

Grovkillen, 2017-12-08:
The lower case issue is a bug that is fixed in current code base. #347

TD-er, 2018-02-19:
This is fixed in PullRequest #891
Also WiFi signal strength is now reported to Domoticz

choxnox, 2018-02-20:
This is great news! Keep up the good work.

choxnox, 2018-02-26:
I had some free time to test this and it works but it's useless in my case. The way battery meter works now is by using internal ADC VCC. This works in cases ESP8266 is directly powered by, let's say, Li-ion battery.

However, I'm currently using Wemos D1 mini + compatible shields and measuring internal ADC VCC is not usable since ESP8266 is powered by 3.3V all the time. The way to measure battery which powers Wemos D1 mini through its compatible battery shield is by connecting battery + terminal on the shield to the A0 via 100k resistor because Wemos already has voltage divider on A0. 

This post sums up things: https://www.letscontrolit.com/forum/viewtopic.php?p=9719#p9719

This post explains how to measure battery for Wemos D1 mini boards: https://arduinodiy.wordpress.com/2016/12/25/monitoring-lipo-battery-voltage-with-wemos-d1-minibattery-shield-and-thingspeak/

I can accurately read correct battery voltage by using `Analog input` task but in that case I'm unable to propagate calculated value (which is in most cases `%value% * 4.2 / 1024`) to Domoticz because current solution reports to Domoticz the value from the internal ADC VCC (which as I said before is meaningless for Wemos D1 mini boards).

TD-er, 2018-02-26:
So we should make a distinction between the internal vcc value and let's say the output of analog input?
I have to think about that and will certainly come back to this.
For now, I will re-open this issue, but set the milestone for it beyond the 2.0 milestone.
It may be a bit harder to implement. Maybe the easiest way would be to let a rule set the reported value of vcc.

uzi18, 2018-02-26:
Just extend "analog input internal" plugin to fill Battery value if Someone name it Battery ;)
And use formula to calculate :)

TD-er, 2018-02-26:
Adding settings to plugins or controllers is a bit tricky, due to the way those are stored in the settings file.

uzi18, 2018-02-26:
So expand api to have plugin_battery_read in plugin...  Or to create event battery#level=something.... And if event occures store it somewhere to use in domoticz plugin

TD-er, 2018-02-26:
That last part was somewhat my suggestion, to allow the reported vcc to be set from within rules. Then you can decide what to send, how to translate it and what sensor to use.

uzi18, 2018-02-26:
This event is even better,  one could use this level in rules to turn on deepsleep until...  Level reach for example 95%

choxnox, 2018-02-26:
I like that idea with rules. This way we can set certain formula because in my case not all Wemos use 100k resistor between VBAT and A0 (some even use 82k to measure battery correctly).

TD-er, 2018-02-26:
But what sets the level?
How do you connect some plugin output to that level? That was what I was talking about.
Adding a trigger to send a report/alarm is also a very good idea.
But the thing with the Domoticz controller is that the battery and signal strength can only be sent along with other values in the same message.
So you cannot connect it to 2 devices, since only one device can send a reply via the Domoticz controller. That's just a limitation in the current design.

uzi18, 2018-02-26:
We can provide variable to be filled if battery event occures for level and controllers could make use with it. If default is -1 so no battery level available

uzi18, 2018-02-26:
In my opinion analog input need to be extended as it is a part of his job and only one input on 8266 we have.
On the other hand system info plugin is also good idea. 
KISS welcome here. 

Note to remember that some input is needed to calculate actual level.

TD-er, 2018-02-26:
And that's why I was thinking about using it in rules. Then you can access the values after formula processing.

choxnox, 2018-02-27:
Is it possible to add another command for rules such as `setDomoticzBatteryValue` which takes value from analog input task and updates internal value/variable which gets sent to Domoticz altogether with sensor data?

TD-er, 2018-02-27:
@choxnox That's what I was suggesting.
But rather use something like `setVccValue` since this value has other use-cases.

choxnox, 2018-02-27:
Yes, naming the command like that makes much more sense and is more universal for other use cases.

choxnox, 2018-02-27:
Actually, now when I gave it a little bit more thought there is an issue in case one uses different battery types. I went through current solution and I see it calculates Domoticz percentage voltage by using hard-coded value of 2.6 inside `mapVccToDomoticz` function.

For example if we take Li-ion batteries, their range of operation can go from 4.2V (100% battery) to let's say 2.8V (~0% battery). This is the reason why `setVccValue` might not be the most accurate solution. Or if it is then there needs to be the way to allow for manual voltage range adjustment depending on the used batteries.

TD-er, 2018-02-27:
True, the Domoticz steps should be mapped also.
But as long as we are considering it as vcc of the ESP, you can add a formula to map the perfect voltage to 3.3V. So if for example your battery pack has 4.2 V as "full", you can map 4.2 => 3.3  (multiply by (33/42)) and then the scale is working again.  
The `setVccValue` could also have 2 parameters:

- measured value
- value considered to be "perfect"

jannnfe, 2018-03-12:
Here is a current workaround for the latest version of ESPEasy?
The code seems to have changed?
I would be very happy if I could send my battery in percent to Domoticz.
http://www.letscontrolit.com/forum/viewtopic.php?f=4&t=1621&p=7317&hilit=battery+status#p7317

TD-er, 2019-10-27:
The battery status is sent when present along with Domoticz controller messages.
So this one can be closed.

closed 2019-10-27 by TD-er