letscontrolit/ESPEasy #569: Feature request: Auto DST

nonflammable, 2017-10-29:
Is it possible to set the daylight saving time automatically ?
I have to change manually twice a year on several devices.

ghtester, 2017-10-29:
Hopefully this obsolete measure will soon be abolished all over the world so this feature won't be necessary anymore...

Grovkillen, 2017-10-29:
Totally agree! A "better" and somewhat more viable option is to be able to update the time offset through commands.

ayasystems, 2017-11-02:
Yes  please!

jjansen85, 2017-11-02:
Maybe we can use this library (https://github.com/JChristensen/Timezone)
I use it in a local project, works very good :)

TD-er, 2017-11-02:
@jjansen85 How do you propose to set the location?
A set of rules with different start- and end-weeks in a combobox + some offset from UTC in another combobox?

papperone, 2017-11-02:
I was in favour to implement this new feature but considering the benefits versus the increase in the code-size I'm not anymore convinced... I think we should evaluate this implication...

TD-er, 2017-11-02:
The suggested library takes about 200 lines of code.
That's not too much, is it?

papperone, 2017-11-02:
on top you need to add "time.h" if not already included in the main ESPEasy code...

papperone, 2017-11-02:
I mean I'm not against it, it's a nice feature, just as it's only 2 times per year, maybe it's more effective a "global" command which can be listened and executed by all ESPEasy module live!
Something like "DSTON" and "DSTOFF"...
Then this can be easily triggered by the server side either via MQTT or HTTP

Grovkillen, 2017-11-02:
Yeah that was my first though as well.

psy0rz, 2017-11-14:
please create an issue on whitehouse.gov and in the EU ;)

mrwee, 2018-01-12:
If ESPEasy is to follow 'regular IT standards' it should support Daylight Saving Time directly and not via MQTT or other command. I have never seen an 'IT product' which only could change DST via an external command. There could be situations where you don't have either MQTT or the possibility to run HTTP commands.

Grovkillen, 2018-01-12:
But still, we have a low budget one-threaded mcu which is having to much free space left. If this is something that needs to be done by one external unit (might still be a ESP Easy unit with special rule to be triggered on dates, or a server) I'd say go for it.

TD-er, 2018-01-12:
I will look into it, to see what the costs in code will be.

---
Verzonden vanaf laptop

mrwee, 2018-01-12:
Thx. Without functional DST, timer based rules running in standalone-mode can be a bit tricky when changing to/from DST. When removing all not-needed plugins there ought to be plenty of space left.

nonflammable, 2018-01-12:
maybe this ?
https://github.com/JChristensen/Timezone

TD-er, 2018-01-14:
I used big chunks of code from the library mentioned (https://github.com/JChristensen/Timezone).
I did not use all of it, since it heavily relied on another Time library, which was already part being used in ESPeasy.
And since that part what was being used, also has some bugs which I fixed earlier (#621) I did not want to go and use it all.
Also that would have increased the size of ESPeasy more than needed.

The currently selected DST will start on Last Sunday March 2am and end on Last Sunday October 3am.
This is the most frequently used DST, according to https://en.wikipedia.org/wiki/Daylight_saving_time_by_country  (and also the one used by most ESPeasy users)

To change the DST definition, currently the code has to be changed in the function applyTimeZone(). Several examples are given in the comments right above that function.
Problem is where to store the information of the timezone (see problems with #710, where only a few bytes were added in the config), how to use it in the web interface and how convenient it is to set by hand.
The UTC offset in minutes still has to be entered and whether or not DST should be enabled.

In later issues, a proper user setup method must be thought of to manually set the DST rules.

closed 2018-03-04 by TD-er