letscontrolit/ESPEasy #573: Feature request: Configurable HTTP webserver port

ghtester, 2017-10-30:
I think it would be nice if the default HTTP port could be easily configurable in the webserver settings please. AFAIK, currently it is "hardcoded" in source file ESPEasy.ino : 

// WebServer
ESP8266WebServer WebServer(80);

As I would like to get several ESPEasy devices temporarily accessible behind my old router (which can support only 1:1 port forwarding), it's not easily possible when all ESPEasy webservers can listen only on the same port 80. 

By the way, thanks a lot to all contributors of this great project. It's a really good work!

TD-er, 2017-10-30:
Another option can be to use apache on a Linux host and let Apache do the "port translation" via a proxypass directive.
Then you can define a virtualhost based on either a domainname, or a portnumber and in such a virtualhost you can let Apache do the re-route to another host.

e.g. bla1.mydomain.com and bla2.mydomain.com can all be pointing to the same A-record of your router and all use port 80. Apache will then route to the proper host internally, based on the domainname in your browser's URL.

LeeNX, 2017-10-30:
I have done the hidden some devices behind a proxy, but there are a few URL that are a few Absolute URLs in the ESPEasy firmware - https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3628

TD-er, 2017-10-30:
I was thinking of another advantage of using Apache as a proxy.
You can add .htaccess to ask for credentials first when someone is trying to access.

You simply do not want ESPeasy to be accessible via the internet without some minimal protection.

@LeeNX how can the URLs be absolute (with included hostname)?

LeeNX, 2017-10-30:
I have secured a domain with an SSL and used OAuth for access, using a single domain, with sub-URL for different devices.  
  
https://espeasy.leenx.lan/ESP-01/ and https://espeasy.leenx.lan/ESP-02/ ... but if you click something like reboot, the URL looks like https://espeasy.leenx.lan/reboot, instead of https://espeasy.leenx.lan/ESP-01/reboot.

TD-er, 2017-10-30:
Ah, I get it now.
That some URLs are relative and some are not, sounds like a bug. Perhaps add a new issue about that?
This issue (http port number) is more like a feature request, but already a work-around is possible using Apache, which in my opinion makes administration a lot easier.

TungstenE2, 2018-06-14:
any news on this? HTTP port setting would be a nice feature.

stevewrightnz, 2019-02-05:
It would also be nice to be able to set the web page username - a bit more secure that just "admin"

uzi18, 2020-05-05:
@LeeNX @TungstenE2 @ghtester so if you want to test it maybe we can ask @TD-er to provide test builds :)

ghtester, 2020-05-06:
@uzi18 A great job, thanks a lot, just quickly tested a fresh test build made by @TD-er and seems to be working fine. :-) I believe we can close this thread as soon as it's merged.

uzi18, 2020-05-06:
I'm glad that you are happy :)

ghtester, 2020-05-26:
Thanks again for a great work, this thread can be closed. :-)

closed 2020-05-26 by ghtester