letscontrolit/ESPEasy #574: Remove absolute or check for X-Forwarded-URL?

LeeNX, 2017-10-30:
I have secured a domain with an SSL and used OAuth for access, using a single domain, with sub-URL for different devices - https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3628

https://espeasy.leenx.lan/ESP-01/ and https://espeasy.leenx.lan/ESP-02/ ... but if you click something like reboot, the URL looks like https://espeasy.leenx.lan/reboot, instead of https://espeasy.leenx.lan/ESP-01/reboot - which breaks.  
  
Either remove or fix all absolute URLs to relative, or check for an HTTP Header like X-Forwarded-URL, which could be added by the proxy service for the base URL and build an absolute URL from the X-Forwarded-URL.

Grovkillen, 2017-10-30:
That sounds like a good idea. +1

psy0rz, 2017-11-14:
yeah we should make all urls relative and consistent

LeeNX, 2017-11-14:
Anything I can help with?

Grovkillen, 2017-11-14:
Yes, feel free to make changes to the code and make a PR

LeeNX, 2017-11-14:
I meant as a non-programer ... ;-)  
  
Maybe a list of the URL that I have noticed with absolute path/URL or something. Maybe testing?

Grovkillen, 2017-11-14:
Yes that is good help as well! :)

psy0rz, 2017-11-15:
a list would be a helpfull start indeed. thanks :)

LeeNX, 2017-11-16:
Quick look through a new device, so might not be complete ...  
  
From the Tools menu:  
 - System: Reboot button - ```/?cmd=reboot```
 - Wifi: Connect - ```/?cmd=wificonnect```
 - Wifi: Disconnect - ```/?cmd=wifidisconnect```  
  
Mmm, was thinking it was more. Also only looks like buttons using ```cmd``` query.  
  
If I find any more URLs, I will post here.

closed 2019-10-11 by Grovkillen