letscontrolit/ESPEasy #575: Feature request: Configurable PwmFrequency and  PwmRange

kianusch, 2017-11-01:
Currently pwm frequency and pwm range hardcoded values (frequency 1000) and (range 1023).

Would be greate to be able to make those values configurable...

e.g. in _P001_Switch.ino ...

        if (command == F("pwmfreq"))
        {
          success = true;

          analogWriteFreq(event->Par1);
          log = String(F("SW   : PWM FREQ ")) + String(F(" Set to ")) + String(event->Par1);
          addLog(LOG_LEVEL_INFO, log);
        }

        if (command == F("pwmrange"))
        {
          success = true;

          analogWriteRange(event->Par1);
          log = String(F("SW   : PWM RANGE ")) + String(F(" Set to ")) + String(event->Par1);
          addLog(LOG_LEVEL_INFO, log);
        }

Thanx

tarkan123, 2018-01-15:
I did find the 'feature' helpful in investigating using the PWM for a specific project - it taught me that the Arduino library is not very good for this !!!  (Tone of 25000hz, only manages ~22500Hz)  :(

cr3pt, 2018-01-23:
Can really help with flickering LED

sebastianheyn, 2018-07-29:
Yeah, this feature would be really helpfull.. I'm using espeasy to dim LEDs with a builtin driver. However depending on the manufacture date they used different driver ICs, one requiring 100-1khz, the other one 5-10khz pwm frequency. Beeing able to set this up in the web frontend would really be awesome!

SchnellerReifen, 2018-08-10:
Would be great, need this for controlling the speed of stepper motors...

closed 2019-10-11 by Grovkillen