letscontrolit/ESPEasy #579: [Feature Request] Add support to home-assistant

leobel96, 2017-11-10:
Hello. Please add support for home-assistant (https://home-assistant.io/): it is improving a lot and becoming more and more stable and featureful and, in my opinion, much better than Domoticz or openHab.

jinjeul, 2017-11-12:
Using MQTT you can link ESPEasy to HA easily.

I'm using it to control ESP relays from Home Assistant.

leobel96, 2017-11-12:
Why there isn't a Home Assistant MQTT option as for Domoticz and openHab?

Grovkillen, 2017-11-14:
You can use OpenHAB's MQTT controller... but yes a GENERIC MQTT controller is more logical.

leobel96, 2017-11-14:
Yes, I think it's more logical.

psy0rz, 2017-11-15:
if its really compatible, maybe we should change the name to OpenHAB/Home assistant?

leobel96, 2017-11-15:
Maybe yes

TD-er, 2017-12-19:
What is the status of this issue? Does the OpenHAB MQTT controller work well with HomeAssistant?

leobel96, 2017-12-19:
Yes works but only if you use mosquitto instead of the embedded mqtt

ZamTheman, 2017-12-23:
Yes I can confirm this. Using mosquitto together with Hassbian.

psy0rz, 2017-12-28:
ok, so perhaps we should update wiki and change the name?