letscontrolit/ESPEasy #580: Feature request:  TSL2591 Digital Light Sensor support

ghtester, 2017-11-11:
Would it be possible to add the TSL2591 sensor please? It's significantly better than supported TSL2561 but unfortunately it's not backward compatible. 
Thanks to all contributors for their great work on this amazing project.
Some resources I have found:
https://learn.adafruit.com/adafruit-tsl2591/wiring-and-test
https://www.letscontrolit.com/forum/viewtopic.php?t=218

krikk, 2017-11-14:
nice sensor, ordered one of these... so if no one writes this in the next 4-6 weeks, i will do it (despite of being a bit short in time for such projects)...

TD-er, 2017-11-14:
I couldn't find any version other than the Adafruit version. I find it hard to believe no-one else made a board for it.
Is there another version (read: cheaper) for this sensor?
Buying one is not an issue, but my sensor projects hardly ever stay at single versions :)

ghtester, 2017-11-14:
Well, I have grabbed it from a bad Samsung smart TV - there's a nice module, TSL2591 combined with IR sensor. I have seen TSL2591 also in some LG TV, on small board with IR sensor as well.

krikk, 2017-11-15:
@TD-er you can find it on aliexpress also, but it is also expensive (9 EUR, with shipping), its a clone of the adafruit board

ShardanX, 2017-12-24:
Meanwhile there is a second TSL2591 on AliExpress that is no adafruit clone judging by another PCB layout.
https://www.aliexpress.com/item/TSL2591-High-Dynamic-Range-Digital-Light-Sensor-Light-module/32309649754.html?ws_ab_test=searchweb0_0,searchweb201602_1_10320_10152_10321_10065_10151_10344_10068_10345_10342_10547_10343_51102_10322_10340_10341_10548_10193_10194_10609_10541_10084_10083_10304_10307_10302_10180_10312_10059_10313_10314_10184_10534_100031_10319_10604_10103_10605_10186_10594_10142_10107,searchweb201603_32,ppcSwitch_5&algo_expid=a33ad52a-b4f9-45c5-91bf-8a61e28e68ea-1&algo_pvid=a33ad52a-b4f9-45c5-91bf-8a61e28e68ea&rmStoreLevelAB=5

psy0rz, 2017-12-31:
thanks krikk! awesome :)

did you receive the senseair sensor from remco already @krikk ?

closed 2017-12-31 by psy0rz

ghtester, 2018-01-02:
Hello, thanks a lot for your great work, could you please compile the firmware binary with this plugin as well? The dev13 binary released today unfortunately does not contain it yet (_P074_TSL2591.ino).

thymjan, 2018-01-02:
Hi five for your implementation! Thank you! Tested the sensor and it works well. Unfortunately I didn't got a connect to my fhem installation. Any hints?
A suggestion: All Light sensors have the same i2c-adress. Maybe a note in the plugin about this?

psy0rz, 2018-01-02:
@ghtester did you use the dev version? (not the test or normal versions)

ghtester, 2018-01-02:
@psy0rz Yes I have installed the first binary release from the .zip file which contained only dev firmware. When you updated the .zip file (so there are also test and normal versions), I opened the src folder and did not find a source codes of _P074 plugin there so I supposed the binary also does not contain the mentioned code.

psy0rz, 2018-01-02:
Ah right..that plugin not in the v2.0-branch, so you need to compile it from the mega-branch yourself.

I'll see if I can setup daily automated releases so that its easier to test for people.

ghtester, 2018-01-02:
OK, thank you, I'm not currently able to compile the firmware by myself but it's not important at all, I'll wait until it's available. Thanks again to you and all contributors for this great work!

psy0rz, 2018-01-02:
too late..am already fixing it. ;)  (i think it benefits all users)

ghtester, 2018-01-02:
Well, I have to say you are awesome... :-) Many thanks again, I am sure you are right.

ghtester, 2018-01-05:
After a quite long testing unfortunately I still can't get the sensor working (as described here: https://github.com/letscontrolit/ESPEasy/pull/661 )
I don't know if it's a HW issue on my side but at least in I2C scan TSL2591 is not listed as supported device on 0x29 address (only TSL2561 is there). I don't know if it may have any impact to sensor initialization or not.
Tested on 2 different NodeMCU Amica boards, one with fresh firmware 20180104, default configuration, only TSL2591 connected. Then played also with **I2C ClockStretchLimit** parameter but without any effect.

TD-er, 2018-01-05:
Do you have pull-up resistors on the SDA and SCL lines?
Some I2C devices really need them.
10kOhm is often enough for 100 kHz I2C speed.

ghtester, 2018-01-05:
Yes I tried 10k pull-ups, then even 1,5k without any effect. I don't know how could I test the I2C communication with specific device (I2C LCD display was working fine on the same ESP module).

TD-er, 2018-01-05:
Did you try to swap SDA and SCL?
Wouldn't be the first cheap China sensor board with incorrect silkscreen text on the PCB.
Try to follow the copper traces to see which pins of the chip are being used.

ghtester, 2018-01-05:
Well, thanks for the hints but the device is detected on 0x29 address (but listed as TSL2561) by I2C scan so I believe SDA and SCL are connected properly. I'm not able to analyze the source code so I don't know how the device detection and initialization works in fact (if it does not see the TSL2591 as supported device).

TD-er, 2018-01-05:
Ah, OK, I missed that part (or misinterpret it) of some detection.
The listed part numbers are only a lookup table which probably has to be updated to list the device.
This is a lookup table which outputs a fixed string of supported devices which may or may not operate on a given address. Each I2C chip only may operate on a fixed (limited) set of addresses, defined by the vendor.
If some device reacts to scan commands on the I2C bus, that address will then be used to lookup in the list of known descriptions.

ghtester, 2018-01-05:
Yes that's my understanding. But I don't know if the TSL2591 plugin can initialize the device properly even though only TSL2561 is in the devices table.

hme0354, 2020-07-27:
Is there something new on this topic? I have an ESP8266 and want to use it with a TSL2591

Grovkillen, 2020-07-27:
https://espeasy.readthedocs.io/en/latest/Plugin/P015_TSL2561.html#p015-tsl2561-page

hme0354, 2020-07-27:
I don't have a TSL2561. I am having a ESL2591

Grovkillen, 2020-07-27:
Ah, my bad. Sorry.

TD-er, 2020-07-27:
And have you tried P074? https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P074_TSL2591.ino

hme0354, 2020-07-28:
how does that work? I only flashed my esp32 with the .bin file for my controller.

TD-er, 2020-07-28:
The plugin is included in the "test" builds.

nikmaier, 2021-06-05:
Hello,
i need you help too, i have an TSL2591.
I have flashed the Test-Build "ESP_Easy_mega_20210503_test_D_ESP8266_4M1M.bin".
But there is no TSL2591 in the device-list.
And when i do an i2c scan, there is only adress 0x29 in use, but no supported device is shown.

What can i do, to run the TSL2591 Lightsensor?

Thanks a lot.

TD-er, 2021-06-05:
> Why is the TSL2591 only in the Test-Builds? i find it works good :-)

It is more of a legacy in naming.
It used to be:
- "normal" -> Stable
- "testing" -> Beta
- "dev" -> It compiles, good luck with it.

But now it is more of a build size issue that's why we need to split it.
Not yet found a good naming schema for it.