letscontrolit/ESPEasy #581: LCD command crashes ESP unit

manjh, 2017-11-11:
### Steps to reproduce
Find an ESP unit with no LCD attached, and send it an LCD command.
Example: "LCD,1,1,this is a test"

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
yes
### Expected behavior
Command should be ignored or rejected

### Actual behavior
ESP crashes and reboots

### System configuration

Hardware: NodeMCU

Software or git version: build 147 (but also happens with older builds)

TD-er, 2017-11-11:
Perhaps also add the forum thread, since that contains a lot more relevant information: https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3762&view=unread#unread

papperone, 2017-11-12:
Looks like an easy fix; it seems the routines to update/refresh LCD are not checking if the "lcd" object has been initialized thus thay coudl create this phenomena...
Not too much free time those days but if nobody is avail I can try to fix it asap I'll have some spare time.

TD-er, 2017-11-12:
Just added a quick fix to prevent dereferencing a null pointer when calling commands to display texts on an uninitialized display.
Did not test it myself, since I do not have these displays.
So please test this pull request #584 to see if it still works when such a display is being used.

I also will try to test it myself to see if it still crashes when no display is attached.

manjh, 2017-11-12:
I can't test it myself right now, since I am away for a short break.

Testing is actually two sided:
1. send a command to an ESP unit that does not have an LCD attached: this
should not reboot.
2. send a command to an ESP unit with LCD attached: this should still work.

When I get back home I can try the second one. The first one can be done
by anyone...

Hans

TD-er, 2017-11-12:
Do you have an example link/command which would fail?

manjh, 2017-11-12:
Simple:
lcd,1,1,testmessage

TD-er, 2017-11-12:
I tried to get one of my sensors to reboot, but it didn't.
It had still a version of last night, so without my fix.

Maybe I do something wrong? I try to call this link: http://192.168.1.89/?cmd=lcd,1,1,blaat
Both with and without the LCD device enabled and/or present, makes no difference.

manjh, 2017-11-12:
That's weird. I was using build 147, and it kept rebooting because of the
command in the system-boot section of the rules.
When I removed that, I was able to get the reboot just by sending a command!
Let me see if I can VPN into my network and retry..

Hans

manjh, 2017-11-12:
OK, just tried: via VPN went to my home network, logged into one of my
Sonoff switches (no LCD), went to the tools page and entered: lcd,1,1,bbbb
Unit rebooted...

Hans

papperone, 2017-11-13:
Quick test on 2 test module and they both reboot (I don't have LCD so I cannot test with one connected)
Later today I'll compile the modified plugin and will try again...

manjh, 2017-11-13:
Good, so I now know I wasn't dreaming...

I can't test at the moment, away for a short brfeak.
Once I am back will be happu to test, I own about 20 ESP units in all
sorts and colors, one has an LCD attached.
It would make things very easy for me if you could provide a loadable
module...

Hans

TD-er, 2017-11-14:
I didn't know what version you need, so I built a few:
https://www.dropbox.com/s/zutl79i8mzowqs1/Builds_ESPeasy_issue581.rar?dl=0
If I'm not mistaken, you'll need the Firmware.bin files. This is just the build directory of my PlatformIO environment and I normally do not upload it myself, just via the PlatformIO module.  :)

manjh, 2017-11-14:
Gijs,

gezien je voornaam moet Nederlands kunnen... toch?
Ik heb de bin files kunnen downloaden, maar heb nog een paar vragen.
Uit de rar filename maak ik op dat dit build 581 is?
Kun je verder uitleggen wat functioneel de verschillen zijn?
Ook herken ik de 1024 en 4096 versies, maar wat is die 8285 versie?

Testen kan ik nu even niet, ik zit tot donderdag in mijn vakantiehuis. Heb
wel een VPB naar thuis, maar ik wil dit soort dingen niet remote
testen.Bovendien is het wat lastig om hiervandaan te zien of het LCD
scherm inderdaad afbeeldt wat ik verwacht....

Ik kan dit weekend wel even spelen denk ik.

groeten,

Hans

TD-er, 2017-11-14:
Ik heb de bugfix gebaseerd op de huidige 2.0 branch. Dus niet de dev-12, maar de aankomende release.

Geen idee waar die buildnr. vandaan komt. Wellicht is dat het aantal builds op mijn laptop????
De 1024 en 4096 versie slaat op de grootte van je flashgeheugen. Dus gelijk aan de versies van de releases die er zijn.
De 8285 is voor versies met niet de ESP8266, maar de afwijkende ESP8285 CPU.

En verder is er nog de 'normal', 'dev' en 'test'. De verschillen slaan op de plugins die erin zitten. Gewoon dezelfde versie gebruiken die je nu ook gebruikt.

For the non-Dutch, switching back to English now :)

psy0rz, 2017-11-15:
merged it and made an additional change. please check if its works now :)

manjh, 2017-11-18:
Gijs,

I have finally been able to complete some tests.
I loaded the normal version into a NodeMCU.
After that I tried a few lcd commands (note that no devices are defined at
this point in time). The commands were rejected, so this is OK.
Then I defined an LCD display, and the commands were now executed.
So I can confirm that the bug has been fixed.

One thing: I tried to upload settings from a file that was created with
build 147. ESPEasy reported that the loading was dons OK, but nothing
showed up. No devices, no rules.
Seems like this is not correct behaviour!

Hans

TD-er, 2017-11-18:
I did not change anything regarding settings.
Was the loading of these settings in the previous build (dev-12) possible?
Perhaps I used an incorrect setting about where the SPIFFS memory segment starts?

manjh, 2017-11-18:
I have no clue. The only observation is the fact that loading settings
from B147 runs OK, but I see nothing coming in.

Hans

ShardanX, 2017-11-18:
Did you try to upload a settings file from R14x to ESP2.0.0?
If i remember right the file format changed from R14x to 2.0.0 so the old version is not loadable to the new one.

manjh, 2017-11-18:
Correct. The backup was created in build147.
If this is true, then it makes transition into the new release very
painful. Personally I have more than 15 units running, and manually
re-configuring all of then will not be trivial.
Is there a possibility for a file conversion utility?

Hans

psy0rz, 2017-11-20:
indeed you cannot use the old config with v2.0 currently. we should probably add a check to the config-uploader so you cant upload incompatible config files.

@mvdbro : would it be easy to convert an old config-file to a new one by just padding the end with zero's? or is the internal structure changed a lot as well?

manjh, 2017-11-20:
If it is possible to detect the difference, it is also an option to create
two load algorithms. This would make the new release compatible with the
previous.

psy0rz, 2017-11-20:
detecting the difference would be no problem, converting it might be quite complicated, depending. will have to look in to the differences.

psy0rz, 2017-11-20:
So can you confirm that everything works correctly if you just use dev12 and reset it to factory defaults and then config it manually?

manjh, 2017-11-20:
Conversion? Why not just two separate routines? One from the old release,
one for the new? I agree it would make the code a little bigger.
A separate conversion utility would not make it bigger, but then the
problem lies in the platform. One routine for the R-Pi, one for Windows,
one for Apple-OS, and so on. I would not go that way.

manjh, 2017-11-20:
Yes. I loaded it into an NodeMCU. Then submitted an LCD command with no
devices attached: command was rejected.
Then defined an LCD device and submitted the command again: worked OK.
So I can confirm this fix has tested OK.

psy0rz, 2017-11-20:
ok, created a new issue for the config incompatibility issues: #595

closed 2017-11-20 by psy0rz