letscontrolit/ESPEasy #583: Dev12 Pulse Counter not working

Grovkillen, 2017-11-11:
I have tried to update my dev7, which count my dehumidifier water extraction, to dev12 but the counter plugin does not work. Downgraded and it started working again.

TD-er, 2017-11-11:
dev7 was released on Mar 15,so that is about the entire history of the P003_Pulse plugin: https://github.com/letscontrolit/ESPEasy/commits/mega/src/_P003_Pulse.ino

When looking at this [commit](https://github.com/letscontrolit/ESPEasy/commit/6de51645b3b28aa2aa4bb42f30ef5bccf821272b#diff-8846b6c63f20a53f51c243807095add8),there was an option added "Mode Type".
The behavior of the old situation was similar to "Falling". That is not the default setting when no value is set.
So perhaps you could test setting that value to "Falling"?

Grovkillen, 2017-11-12:
> mega/src/_P003_Pulse.ino

Thanks for investigating Gijs! Will try!

Grovkillen, 2017-11-27:
I will have time to go over this during the upcoming weekend.

Grovkillen, 2017-12-03:
Note to myself:

https://datasheets.maximintegrated.com/en/ds/DS2423.pdf

1-wire support for DS2423??

TD-er, 2017-12-06:
And? Have you tried set the Mode Type to Falling?

Grovkillen, 2017-12-07:
Sorry, haven't come around to that yet. Will have time for it on Saturday (when I'm back from a trip).

Grovkillen, 2017-12-09:
I have now tested it. The plugin is not counting at all with dev12...  but then I found this in the log:

`PULSE: Error, only the first 4 tasks can be pulse counters.`

I moved the device from task 5 to task 4 and it started to work. Is there a reason for this? I close it and will add information about this on the wiki but it would be nice to state why this is a restriction. Plus that this information need to popup when the plugin is selected.

Grovkillen, 2017-12-09:
I have now added information regarding the restriction to these three pulse counter pages:

https://www.letscontrolit.com/wiki/index.php/TCR5000#ESP_Easy
https://www.letscontrolit.com/wiki/index.php/LJ12A3#ESP_Easy
https://www.letscontrolit.com/wiki/index.php/Pulse_logic_input#ESP_Easy

closed 2017-12-09 by Grovkillen

TBobsin, 2020-01-04:
In this Firmware only task 1 to 3 seems to work. 
Task 4 dosn't count
Firmware
--
Build:⋄ | 20104  - Mega
System Libraries:⋄ | ESP82xx Core 2_6_2, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20191208