letscontrolit/ESPEasy #588: Error while reading/writing config.dat in 926 - (TaskValueSet)

HV-NL, 2017-11-13:
Regularly (but not always) a **reboot** occurs on 2.0.0dev12 (the last line in syslog before such a reboot normally is "TaskValueSet").

The **first** call to TaskValueSet after a reboot will generate "**Error while reading/writing config.dat in 926**". 

### Steps to reproduce
_How can we trigger this problem?_

- create a Dummy Device (I use device 3)
- create a rule
 `On System#Boot do`
  `TaskValueSet 3,1,0 // Zet de 1e param van dummy task #3 op 0`
 `endon`
- manually set the value once (tools/command)
 `TaskValueSet 3,1,1`
- perform a reboot

_Does the problem persist after powering off and on? (just resetting isnt enough sometimes)_

No. Directly after a power-on it does not occur.
But once reboot occurs, then it will always give the above mentioned error (after each subsequent reboot).

### Expected behavior
_Tell us what should happen?_

No error messages "Error while reading/writing config.dat"

### Actual behavior
_Tell us what happens instead?_
See above.

### System configuration

_Hardware:_
NodeMCU

_Software or git version:_
version 2.0.0dev12 (4 Mb version).

_log info:_
 Nov 13 15:17:21 ESP Unit: 1 : ACT  : TaskValueSet 3,1,1
 Nov 13 15:17:36 ESP Unit: 1 : WIFI : Connected! IP: 192.168.11.153 (CV-ketel_1)
 Nov 13 15:17:37 ESP Unit: 1 : EVENT: System#Boot
 Nov 13 15:17:37 ESP Unit: 1 : ACT  : TaskValueSet 3,1,0
 Nov 13 15:17:37 ESP Unit: 1 : FS   : Error while reading/writing config.dat in 926
 Nov 13 15:17:37 ESP Unit: 1 : FS   : Error while reading/writing config.dat in 926
 Nov 13 15:17:37 ESP Unit: 1 : FS   : Error while reading/writing config.dat in 926
 Nov 13 15:17:38 ESP Unit: 1 : FS   : Error while reading/writing config.dat in 926
 Nov 13 15:17:38 ESP Unit: 1 : FS   : Error while reading/writing config.dat in 926
 Nov 13 15:17:38 ESP Unit: 1 : EVENT: Processing time:322 milliSeconds
 Nov 13 15:17:38 ESP Unit: 1 : WD   : Uptime 0 ConnectFailures 0 FreeMem 19024

Grovkillen, 2017-11-14:
More info in the forum thread: https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3570&start=40#p20736

psy0rz, 2017-11-15:
looking into it now, might be an existing bug that came to light because of some extra error checking i've added to load/save functions.

psy0rz, 2017-11-16:
strange..i cannot reproduce it.

taskvalueset should not trigger a config save.

do you have any other rules (check all rule sets)? 

My output:

 ```
EVENT: System#Boot
 ACT  : TaskValueSet 3,1,0  
 EVENT: Processing time:131 milliSeconds

```

psy0rz, 2017-11-16:
where does your first taskvalueset come from, at Nov 13 15:17:21?

HV-NL, 2017-11-16:
> where does your first taskvalueset come from, at Nov 13 15:17:21?

From the rule 'On System#Boot do'
I do have some more rules, but only 1 line is called at boot time.

This is my complete setup:
* device 1: DS18b20
* device 2: DS18b20
* device 3: dummy device

All errors seem to be related to this 3rd (dummy) device. TaskValueSet will either generate a reboot, or 'error while reading/writing'.

This is my complete rule set:

`On System#Boot do`
`  TaskValueSet 3,1,0 // Zet de 1e param van dummy task #3 op 0`
` endon`

`On CV-In#Temperature-In do`
`  if [CV-In#Temperature-In]>30`
`   event InlaatTemperatuurOK`
`  endif`
`  if [CV-In#Temperature-In]<28`
`   event PompKanUit`
`  endif`
` endon`
 
` On InlaatTemperatuurOK do`
`  if [CV-Uit#Temperature-Uit]>35`
`   event PompKanAan`
`  endif`
` endon`
 
` On PompKanAan do`
`  if [PompMagAan#PompMagAan]=0`
`   TaskValueSet 3,1,1 // Zet de 1e param van dummy task #3 op 1`
`  endif`
` endon`
 
` On PompKanUit do`
`  if [PompMagAan#PompMagAan]=1`
`    TaskValueSet 3,1,0 // Zet de 1e param van dummy task #3 op 0`
`  endif`
` endon`

HV-NL, 2017-11-16:
> do you have any other rules (check all rule sets)?

I only have 1 set of rules:
[image]

HV-NL, 2017-11-16:
Just downloaded the current github source code, and made a slight modification to the debug output

`#define SPIFFS_CHECK(result, fname)   if (!(result)) { return(FileError(__LINE__, fname)); }`
`#define SPIFFS_CHECK_W(result, fname) if (!(result)) { return(FileErrorW(__LINE__, fname)); }`
(FileError for read, FileErrorW called for writing to a file, and I created FileErrorW)

Now I can clearly see that the error message is related to reading:

`Nov 16 13:48:24 ESP Unit: 1 : EVENT: System#Boot`
`Nov 16 13:48:24 ESP Unit: 1 : ACT  : TaskValueSet 3,1,0`
`Nov 16 13:48:24 ESP Unit: 1 : FS   : Error while reading config.dat in 969`
`Nov 16 13:48:24 ESP Unit: 1 : FS   : Error while reading config.dat in 969`
`Nov 16 13:48:24 ESP Unit: 1 : FS   : Error while reading config.dat in 969`
`Nov 16 13:48:24 ESP Unit: 1 : FS   : Error while reading config.dat in 969`
`Nov 16 13:48:24 ESP Unit: 1 : FS   : Error while reading config.dat in 969`
`Nov 16 13:48:24 ESP Unit: 1 : WD   : Uptime 0 ConnectFailures 0 FreeMem 20824`

(line number 926 is changed to 969 due to my extra code, but still points to the second SPIFFS_CHECK statement in 'LoadFromFile'):

`/********************************************************************************************\`
`  Load data from config file on SPIFFS`
`  \*********************************************************************************************/`
`String LoadFromFile(char* fname, int index, byte* memAddress, int datasize)`
`{`
`  // addLog(LOG_LEVEL_INFO, String(F("FILE : Load size "))+datasize);`
` `
`  fs::File f = SPIFFS.open(fname, "r+");`
`  SPIFFS_CHECK(f, fname);`
` `
`  // addLog(LOG_LEVEL_INFO, String(F("FILE : File size "))+f.size());`
`  SPIFFS_CHECK(f.seek(index, fs::SeekSet), fname);` **<== THIS LINE**
`  byte *pointerToByteToRead = memAddress;`
`  for (int x = 0; x < datasize; x++)`
`  {`
`    int readres=f.read();`
`   SPIFFS_CHECK(readres >=0, fname);`
`    *pointerToByteToRead = readres;`
`    pointerToByteToRead++;// next byte`
`  }`
`  f.close();`
` `
`  return(String());`
`}`

psy0rz, 2017-11-20:
thanks that helps!

psy0rz, 2017-11-20:
ok, created a new issue for the config incompatibility issues: #595

psy0rz, 2017-11-21:
Did some debugging, but i need to sleep now. :)

It seems config.dat can get read by calling:
LoadSettings
LoadTaskSettings
LoadCustomTaskSettings
LoadControllerSettings
LoadCustomControllerSettings

Some of these can again be called by calling rulesProcessing.

Its probably a bug somewhere that came to light by the extra error checking i've added in loadfromfile.

I'll try to recreate the problem on my side by adding all your devices and rules. I'll also make sure to make error logging or debug logging better so we can see exactly which call triggered the read error. Its most likely that its trying to seek to an illegal position. (Because loadfromfile is called with an incorrect index)

Important we figure this out since it can trigger a lot of weird issues.

psy0rz, 2017-11-22:
ok i can reproduce the problem. i also create automated tests for the ds18b20 to speed up future debugging.

psy0rz, 2017-11-22:
ok please check commit a1fc351 to see if its fixed now. it was indeed a bug.

closed 2017-11-22 by psy0rz

psy0rz, 2017-11-22:
also merged in 2.0

commit a1fc3511 (2017-11-22): fix #588. parseTemplate() now only checks enabled taskes for [] macros
commit 64caeac9 (2017-11-22): fix #588. parseTemplate() now only checks enabled taskes for [] macros