letscontrolit/ESPEasy #589: Slow wifi reconnecting

melwinek, 2017-11-15:
When I turn off my home wifi when the ESP is connected. Every now and then, ESP tries to connect to wifi. This causes a slight slowdown of work. This is not a problem.

When I turn on ESP when WIFI is off.
Reading the status of the buttons is delayed for many seconds.
Rule action also works very slowly.
Using ESPEASY is practically impossible.

Grovkillen, 2017-11-15:
I can confirm this. The same goes with lost connection with my MQTT broker.

beicnet, 2017-11-15:
Same issue here from a long time ago, but did not want to open a ticket for it! :D

psy0rz, 2017-11-15:
Yep, currently ESPEasy is designed to be always connected with the wifi. If its not, it will try and retry, which will take a lot of time.

We can probably make a lot of improvements in this area.

TD-er, 2017-12-09:
Suggestion:
Make controller plugins and MQTTConnect aware of the WiFi status.
It is useless to try to connect these when there is no active WiFi connection.

Is it OK to create a patch for this, based on the #629 pull request? (which means 2.0 branch)
Or should we do that on 2.1?

TD-er, 2017-12-12:
Another usecase, described here: https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3867&p=21526#p21524
The ESP WiFi should be more flexible in making connections and retries.

TD-er, 2017-12-12:
I added a few checks to prevent unneeded timeouts with network attempts when not connected.

It is based on #629 Thus that must be merged first.

TD-er, 2017-12-13:
And 2nd part of this has been added to the pull request #639 
Power-up without the WiFi present now takes about 30 seconds and then every other thing starts working like it should.
All controllers stop sending (and waiting for time-out) when not connected.

As soon as the WiFi connection is restored, a NTP request is performed and all other controllers start working again.

N.B. All data recorded during offline will NOT be kept and re-sent.

TD-er, 2018-01-17:
Did someone test the fix to see if it is good enough with the latest releases?

Grovkillen, 2018-03-16:
We need to discuss if this is considered fixed or not?

melwinek, 2018-03-23:
I think it is very good now.

Oxyandy, 2018-03-23:
Add a notify onboot to a rule that sends an email..
This was working great recently for me 
Then Using ESP_Easy_mega-20180322_normal_ESP8266_1024.bin it stopped working.
If you are using a different version to me, please let me know 'which' and I will confirm it works

TD-er, 2018-03-23:
And with the upcoming change it is even getting better:

```
EVENT: Clock#Time=Fri,22:28
WD   : Uptime 2 ConnectFailures 0 FreeMem 13688
EVENT: WiFi#Disconnected
WIFI : Disconnected! Reason: Beacon timeout Connected for 0h01m
WIFI : Connecting Lurch2 attempt #0
EVENT: WiFi#Disconnected
WIFI : Disconnected! Reason: No AP found Connected for 0h01m
WIFI : Connecting Lurch2 attempt #1
EVENT: WiFi#Disconnected
WIFI : Disconnected! Reason: No AP found Connected for 0h01m
WIFI : Connecting Lurch2 attempt #2
EVENT: WiFi#Disconnected
WIFI : Disconnected! Reason: No AP found Connected for 0h01m

WIFI : Connecting Lurch3 attempt #3
EVENT: Clock#Time=Fri,22:29
WIFI : Connected! AP: Lurch3 AP BSSID: CC:CE:1E:13:B7:58 Ch: 11 took: 3044 ms
EVENT: WiFi#ChangedAccesspoint
WIFI : Connected! IP: 192.168.1.75 (ESP_Easy-0) AP: Lurch3 AP BSSID: CC:CE:1E:13:B7:58 took: 678 ms
EVENT: WiFi#Connected
MQTT : Connection lost
MQTT : Connected to broker with client ID: ESPClient_A0:20:A6:19:2A:BF
Subscribed to: domoticz/out
EVENT: MQTT#Connected
```

This was a test of my new approach where we're not even actively polling on events, but wait for events to happen and act upon them as soon as the background processes notice a change.
In this test the first accesspoint was restarted and thus offline for a while.
Timeouts for WiFi.begin() are stopped the moment the connection attempt gets disconnected at the lowest level.

And this approach also allows for some more information:
[image]

To-be-continued :)

uzi18, 2018-03-23:
Try new events for connection info,  it is possible  no connection at all at boot.

Grovkillen, 2018-03-23:
How come MQTT connection starts with a "Connection lost"?

TD-er, 2018-03-23:
In my log output you mean?
There was a fully functioning situation and then I rebooted the accesspoint so it lost connection.
3 reconnect attempts + successful connect to another accesspoint took about 15 seconds.