letscontrolit/ESPEasy #590: More than 12 tasks

melwinek, 2017-11-15:
Is it possible for you to safely add more than 12 tasks in the next builds.
I know that it is possible to change and recompile the source yourself, but this can cause problems.

Grovkillen, 2017-11-15:
It might be considered in the future but not for now, it is suggested that you break the unit down to multiple ESP-modules as of right now.

hexenmeister, 2017-11-15:
This is possible (up to about 28 without enlargement of config.dat), but some addresses have to be recalculated

I did it that way:
#define DAT_OFFSET_TASKS                 4096  // each task = 2k, (1024 basic + 1024 bytes custom), 12 max
#define DAT_OFFSET_CONTROLLER            DAT_OFFSET_TASKS + (DAT_TASKS_SIZE * TASKS_MAX)                        // each controller = 1k, 4 max
#define DAT_OFFSET_CUSTOM_CONTROLLER     DAT_OFFSET_CONTROLLER + (DAT_CUSTOM_CONTROLLER_SIZE * CONTROLLER_MAX)  // each custom controller config = 1k, 4 max.

see here: https://github.com/hexenmeister/ESPEasy/blob/mega/src/ESPEasy.ino

melwinek, 2017-11-15:
Grovkillen what do you think about what the hexenmeister wrote?
Maybe it will be included in 2.0.0?

Grovkillen, 2017-11-15:
Not in 2.0.0 since we're trying to close the version asap. But I'll add it
to 2.1.0 :)

vikasjee, 2017-11-30:
Any Release Calendar for 2.1.0? Looking forward to it very eagerly...

Grovkillen, 2017-12-01:
I'd say 2.1.0 is some month further into the future. Sorry but we have a bit of work ahead of us with the 2.1.0 (we haven't set a road map or main focus regarding the next version).

Grovkillen, 2017-12-14:
@Psy0rz what do you think about this? I would recommend us switching it up to 24 devices by default. Maybe even as a 2.0.0 milestone?

TD-er, 2017-12-15:
Given issue #595 it might be a good idea to implement more devices in 2.0.
However, it should not use extra memory.
Not sure if adding extra device entries will increase memory usage.
If I remember correctly, some things were declared based on the number of devices.

psy0rz, 2017-12-18:
Good point, since the config in v2.0 already is fscked you mean. :)

Not sure how much of an inpact it would have, but i'm sure it uses extra memory.

Grovkillen, 2017-12-19:
Yeah it will probably use extra memory but the 12 devices limit is a bit of a bummer sometimes. 24 would make it a lot more usable. Would it be possible to scrape the number of active tasks for it to use less memory if less tasks is enabled?

psy0rz, 2017-12-28:
will have to look into it, it will have quite an impact i think.

TD-er, 2017-12-28:
memory as in RAM, or space reserved for settings?

psy0rz, 2017-12-31:
RAM.

vikasjee, 2018-06-18:
When is milestone 2.2 "KNOX" expected to be released? Its been too long...

uzi18, 2018-10-23:
@itProfi I have working configuration for 24 tasks, have mentioned in different issue

ceca0105, 2019-01-18:
@uzi18 could you send me your compiled 24 tasks version firmware?

uzi18, 2019-01-18:
@ceca0105 what kind of build you need? need to warn you as configuration is incompatible with official build, so you need to manually reconfigure unit.

tparvais, 2020-02-04:
@uzi18 I also tried to compile ESPEasy_mega-20191208.zip with tasks_max = 20
I observed sometimes on Wemos D1 mini, after wifi AP config, the Wemos cycles during boot with the following:

any idea ?

`load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
¬U87 : Info  :

INIT : Booting version: (custom) (ESP82xx Core 2_6_3, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2)
88 : Info  : INIT : Free RAM:36192
89 : Info  : INIT : Warm boot #538 Last Task: Background Task - Restart Reason: Exception
91 : Info  : FS   : Mounting...
140 : Info  : FS   : Mount successful, used 75802 bytes of 1953282
148 : Info  : CRC  : No program memory checksum found. Check output of crc2.py
158 : Info  : CRC  : SecuritySettings CRC   ...OK
370 : Info  : INIT : Free RAM:33056
371 : Info  : INIT : I2C
372 : Info  : INIT : SPI not enabled
461 : Info  : INFO : Plugins: 46 [Normal] (ESP82xx Core 2_6_3, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2)
565 : Info  : WIFI : Set WiFi to STA
599 : Info  : WIFI  : Start network scan
600 : Info  : WIFI : Connecting Orange-70a attempt #0
Exception (9):
epc1=0x40275724 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff0f96 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd40 end: 3fffffc0 offset: 01a0
3ffffee0:  0000000e 3fff4c6c 3fff03cc 3fff0c44
3ffffef0:  3fff0c24 3fff0c04 00000020 4010168b
3fffff00:  3fff0c24 3fff4c6c 00000002 40223a28
3fffff10:  40276030 00000000 40276030 00000000
3fffff20:  40276030 00000000 40276030 00000000
3fffff30:  3fff0c24 3fff0c04 3fff0170 40237c59
3fffff40:  3fff4c00 0027002f 80ff0f8c 40268f4c
3fffff50:  2e310029 3ffe865c 00000002 3fff18dc
3fffff60:  3fff03c9 3ffe865c 3fff0f8c 4024e59e
3fffff70:  3fff4000 003d003f 8027005d 40273ed4
3fffff80:  3fff324c 005b006f 00efeffe feefeffe
3fffff90:  feefeffe feefeffe feefeffe 3fff1a0c
3fffffa0:  3fffdad0 00000000 3fff19cc 4026a16c
3fffffb0:  feefeffe feefeffe 3ffe86e0 4010194d
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)`

TD-er, 2020-02-04:
We only tested it with 24, so I guess a lot will go wrong with 20 tasks as max.

tparvais, 2020-02-04:
> 
> 
> We only tested it with 24, so I guess a lot will go wrong with 20 tasks as max.

so if I use 24 to recompile it will work ? with 4MB and FS=1MB ?

TD-er, 2020-02-04:
Where exactly have you made the change?

tparvais, 2020-02-04:
in ESPEasyLimits.h, I changed 12 to 16 or 20
Then I just discover a DEFINE in custom.h
It's compiling

tparvais, 2020-02-04:
mmmh, it does compile with standard 12 tasks.
I connected to my Wifi
Then I tried to fix IP@, and reboot and paf ! it cycles rebooting ! with same debug info above

so this is not related to Max_Tasks

TD-er, 2020-02-04:
Fixed IP is an issue that will be fixed as soon as I merge my PR https://github.com/letscontrolit/ESPEasy/pull/2864

Which I will merge right now.