letscontrolit/ESPEasy #592: wrong systime with DST / Timezone enabled

marcohald, 2017-11-18:
### Steps to reproduce
Enable Rule1 with this Script https://github.com/hphde/espeasy-rules/blob/master/timer-switch-sonoff-s20-rule.txt
Enable the NTP Server and configure  Timezone Offset 60, DST or Timezone Offset 0, DST on.
Plug in between 22:01 and 22:59.

### Does the problem persist after powering off and on? 
Yes

### Expected behavior
Relay should stay off

### Actual behavior
Relay is on

### System configuration

Hardware:
Sonoff S20
Software or git version:
ESPEasy_v2.0.0-dev12_normal_1024.bin

### Explanation
When I plug in the Socket before the off Time 22:00 is reached the Rule works a  intended
But when I plug in the Socket between  22:00 and 23:00 the relay powers on (should stay off).
After 23:00 the relay stays off as intended.
I think Clock#Time honors the DST / Timezone and %systime% not

TD-er, 2017-12-05:
I was looking into these rules and wonder if this ever will work:
` if %systime% < 12:00`
%systime% will expand to "22:33:44" or something like that.

The compare is done via a "toFloat()" function and I am not sure what will actually be compared.
It may depend on a leading space (e.g. " 12") and interpreting of leading zeros, which may be considered being octal number or just "NaN".
Compares with a "NaN" are always tricky.

Also, what if the NTP update was not (yet) performed, then the systime does not have the DST applied. (see now() in Misc.ino)
Systime is then something like 00:00:00 (give or take a few seconds)

So it might be the compare function never works when comparing with times before 10am and perhaps the timeout of the NTP is also too short.

TD-er, 2018-01-02:
This could also have been fixed with other pull requests like #639 in dev-13.
In that pull request the NTP update is being done immediately after WiFi connection is made.
This may (or may not) fix this issue.
So please test if it is fixed now, or just made harder to reproduce ;)

TD-er, 2018-01-29:
@marcohald Is it fixed now?

TD-er, 2018-02-10:
Hmm @vader wrote me a personal message on the forum, about something similar, related to the if %systime% < ... 
So maybe it is related to this and then I didn't break something with a recent PR ;)

psy0rz, 2018-02-26:
should be fixed, please reopen if this problem still exists.

closed 2018-02-26 by psy0rz

uzi18, 2018-02-26:
For me String("12:34").toFloat() == 12.0, need to test it ...

TD-er, 2018-02-26:
But that's another issue and probably nothing to do with DST, right?
There is something fishy in the rule handling.
It doesn't look for times and tries to handle it like int or float, which obviously doesn't work like that.

uzi18, 2018-02-26:
Here You are:
Rules:   

```
on oldtime do
if %systime%>12:34
event ok
else
event nok
endif
endon

on newtime do
if %systime%>01:23:45
event ok
else
event nok
endif
endon

```
Log:   
```
>>Send to /dev/ttyUSB1: "event oldtime"<<
>event oldtime
EVENT: oldtime
to check: 23:03:39>12:34
12:34 String -> Float 12.00
23:03:39 String -> Float 23.00
compare: > match
ACT  : event ok
EVENT: ok

Ok

>>Send to /dev/ttyUSB1: "event newtime"<<
>event newtime
EVENT: newtime
to check: 23:03:52>01:23:45
01:23:45 String -> Float 1.00
23:03:52 String -> Float 23.00
compare: > match
ACT  : event ok
EVENT: ok

Ok
```

uzi18, 2018-02-26:
So it is totally wrong here

uzi18, 2018-02-26:
Have no idea how to aproch it, calculate seconds and then compare?
We can also just remove chars ':', simplest way
@TD-er what do You think?

TD-er, 2018-02-26:
Remove ":" is one approach. Not sure if that character is also used elsewhere in the rules.

Another approach is to add a simple check on the "Number" if it is a valid float (I recently added a check for that in the code, isFloat() ) and if not, try to figure out what it is.

Simply removing the ":" can lead to strange results with leading zero's. Those can be considered octal representation.

uzi18, 2018-02-26:
Better to convert to seconds and then compare is easy

TD-er, 2018-02-26:
Indeed, but better use the isFloat() function to detect whether the characters in the substring represent a float or int.
If not, then first consider it a time notation and convert it to seconds or minutes.
There is already some "BCD"-like approach for time conversion in rules.