letscontrolit/ESPEasy #593: Device number > 3 does not have Name

wolverinevn, 2017-11-20:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Add 3 devices (real devices or dummy)
From 4th device, Esp won’t save the name of device or value. Submit, back to Device list, just blank name and value name.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
Tell us what should happen?
Device name and Value name

### Actual behavior
Tell us what happens instead?
Name was not saved

### System configuration

Hardware: NodeMCU Esp-12

Software or git version: dev beta 12

psy0rz, 2017-11-20:
Its actually dev-12, we're not in beta-mode yet. will look into it.

wolverinevn, 2017-11-21:
My bad. After few reboot, it works as inteded now.

closed 2017-11-21 by wolverinevn

psy0rz, 2017-11-21:
ok thanks