letscontrolit/ESPEasy #594: Devices (sensors) & NTP seem to crash after 49.7 days uptime

TD-er, 2017-11-20:
### Steps to reproduce
How can we trigger this problem?

Leave it up for > 49 days :)
N.B. probably related to some 32 bit millisecond timer which overflows after 49.7 days.
But the internal timer was only a few hours before that point. (believe it or not, the last log was at 71337 minutes uptime)
Maybe it is related to computing the time diff based on first connection?

Does the problem persist after powering off and on? (just resetting isn't enough sometimes)
Probably not.

### Expected behavior
Tell us what should happen?

Things should continue to work.

### Actual behavior
Tell us what happens instead?

NTP remains frozen at the moment  the last value is recorded in Domoticz.
Sensor values are no longer updated.
Only the uptime is being logged.
WiFi remains connected and logs are still recorded to the log server.

When disable/enable NTP, the displayed time is only updated once.
Web page remains responsive.

Disable/Enable devices does not make them work again.
I2C communication works, since I can change the I2C address of the BMx280 device and it detects the device properly.

### System configuration

Hardware:

- NodeMCU 1.0
- MHz-19 CO2 sensor
- BMP280

Software or git version:
2.00 custom build version (endurance test of the plugins I changed myself)

Last log entry:
```
Nov 20 05:16:22 ESP Unit: 1 : MQTT : Topic: domoticz/out
Nov 20 05:16:22 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "General",#012   "id" : "00082078",#012   "idx" : 78,#012   "meterType" : "Energy",#012   "name" : "defect CO2_Mqtt",#012   "nvalue" : 0,#012   "stype" : "Custom Sens
Nov 20 05:16:22 ESP Unit: 1 : WD   : Uptime 71337 ConnectFailures 0 FreeMem 21608
Nov 20 05:16:22 ESP Unit: 1 : UDP  : Send Sysinfo message
Nov 20 05:16:23 ESP Unit: 1 : BMP280: Apply temp offset -9.30C temperature 27.25C => 17.95C
Nov 20 05:16:23 ESP Unit: 1 : BMP280 : Address: 0x76
Nov 20 05:16:23 ESP Unit: 1 : BMP280 : Temperature: 17.95
Nov 20 05:16:23 ESP Unit: 1 : BMP280 : Barometric Pressure: 1017.47
Nov 20 05:16:23 ESP Unit: 1 : MQTT : {"idx":72,"nvalue":0,"svalue":"17.95;0.00;0;1017.47;0"}
Nov 20 05:16:23 ESP Unit: 1 : MHZ19: PPM value: 413 Temp/S/U values: 23/64/10345.00
Nov 20 05:16:23 ESP Unit: 1 : MQTT : {"idx":78,"nvalue":0,"svalue":"413.00;23.00;10345.00"}
Nov 20 05:16:23 ESP Unit: 1 : HTTP : connecting to 192.168.1.74:8080
Nov 20 05:16:23 ESP Unit: 1 : HTTP/1.1 200 OK#015
Nov 20 05:16:23 ESP Unit: 1 : HTTP : Success
Nov 20 05:16:23 ESP Unit: 1 : Content-Length: 53#015
Nov 20 05:16:23 ESP Unit: 1 : Content-Type: application/json;charset=UTF-8#015
Nov 20 05:16:23 ESP Unit: 1 : Cache-Control: no-cache#015
Nov 20 05:16:23 ESP Unit: 1 : Pragma: no-cache#015
Nov 20 05:16:23 ESP Unit: 1 : Access-Control-Allow-Origin: *#015
Nov 20 05:16:23 ESP Unit: 1 : #015
Nov 20 05:16:23 ESP Unit: 1 : {
Nov 20 05:16:23 ESP Unit: 1 :    "status" : "OK",
Nov 20 05:16:23 ESP Unit: 1 :    "title" : "Update Device"
Nov 20 05:16:23 ESP Unit: 1 : }
Nov 20 05:16:23 ESP Unit: 1 : HTTP : closing connection
```
The logs on the logserver show only messages indicating activity of other sensors like this:
```
Nov 20 05:16:23 ESP Unit: 1 : MQTT : Topic: domoticz/out
Nov 20 05:16:23 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "General",#012   "id" : "00082071",#012   "idx" : 71,#012   "meterType" : "Energy",#012   "name" : "defect  CO2",#012   "nvalue" : 0,#012   "stype" : "Custom Sensor",
Nov 20 05:16:23 ESP Unit: 1 : MQTT : Topic: domoticz/out
Nov 20 05:16:23 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "YouLess Meter",#012   "id" : "80",#012   "idx" : 43,#012   "meterType" : "Energy",#012   "name" : "Unknown",#012   "nvalue" : 0,#012   "stype" : "YouLess counter",#012
Nov 20 05:16:23 ESP Unit: 1 : MQTT : Topic: domoticz/out
Nov 20 05:16:23 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "General",#012   "id" : "00082078",#012   "idx" : 78,#012   "meterType" : "Energy",#012   "name" : "defect CO2_Mqtt",#012   "nvalue" : 0,#012   "stype" : "Custom Sens
Nov 20 05:16:23 ESP Unit: 1 : MQTT : Topic: domoticz/out
Nov 20 05:16:23 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "General",#012   "id" : "00082071",#012   "idx" : 71,#012   "meterType" : "Energy",#012   "name" : "defect  CO2",#012   "nvalue" : 0,#012   "stype" : "Custom Sensor",
Nov 20 05:16:24 ESP Unit: 1 : MQTT : Topic: domoticz/out
Nov 20 05:16:24 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "Temp + Baro",#012   "id" : "82080",#012   "idx" : 80,#012   "meterType" : "Energy",#012   "name" : "breadboard bmp280",#012   "nvalue" : 0,#012   "stype" : "BMP085 I
Nov 20 05:16:32 ESP Unit: 1 : MQTT : Topic: domoticz/out
Nov 20 05:16:32 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "YouLess Meter",#012   "id" : "80",#012   "idx" : 43,#012   "meterType" : "Energy",#012   "name" : "Unknown",#012   "nvalue" : 0,#012   "stype" : "YouLess counter",#012
Nov 20 05:16:38 ESP Unit: 1 : MQTT : Topic: domoticz/out
Nov 20 05:16:38 ESP Unit: 1 : MQTT : Payload: {#012   "Battery" : 255,#012   "RSSI" : 12,#012   "description" : "",#012   "dtype" : "General",#012   "id" : "00082073",#012   "idx" : 73,#012   "meterType" : "Energy",#012   "name" : "ESPeasy CO2",#012   "nvalue" : 0,#012   "stype" : "Custom Sensor",
Nov 20 05:16:39 ESP Unit: 1 : UDP  : a0:20:a6:19:2b:28,192.168.1.80,3

```

Logs in the web-interface only show lines like these:
`53190999 : WD : Uptime 72224 ConnectFailures 0 FreeMem 21352
`
And of course changes in settings, but no signs of sensor activity.

[image]

psy0rz, 2017-11-20:
I just checked a esp that wasnt functioning correctly: It was up longer than 49 days, the DS18b20 sensors still where sending updates to domoticz, but the simple Switch input was not responding to input changes. not sure if its related, will compare ds18b20 to Switch input to see if my problem also was time related.

Will add it to 2.0 but if it proves too difficult to fix i will bump it to 2.1.0. Hopefully we can trigger the bug directly via hack, instead of waiting 49 days each time ;)

TD-er, 2017-11-23:
Topic from other user with similar problems on the forum: https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3803&p=21008#p21008

papperone, 2017-11-23:
I can confirm this reoccurrence (today 3rd time so it's a pattern for me) but only for one module...
This module has 2xRelay (as inverted switch), 2xLEDs (to show if realy is on or off) and a DS18B20.
The phenomena on my module is that relays works perfectly when activated with Events sent via MQTT, the sensors reads temperature without issues but the LED stays in the last known state :(
(this is a unit which control my garden lights with a timer) this early morning when the last set of light turned off perfectly the relevant LED stayed ON; I try to operate both relay via MQTT and all work perfectly , just the 2 LEDs were not responding.
Setup: Relay = GPIO12 and 24 - LEDs = GPIO 4 and 5, DS18B20 = GPIO0

CherAlban, 2017-11-24:
I can also confirm this bug for all of my Sonoff S20s, they stop acting, while the web-interface is still reachable, some get back to normal with a soft reboot, others need power cycling to get back to work.

papperone, 2017-11-24:
I try to investigate further and definitely somethign has to do with the way devices are handled.
In my settings (as per above) I've devices defined for LEDs and DS18B20 but not for the relays which are triggered by GPIOs command in the rules. after the 49+ days the relays are still responsive to GPIOs commands (either directly from the WEB-GUI and MQTT) but the LEDs are not (even with direct GPIOs commands); just the DS18B20 despite being a device it keeps sending his reading trough MQTT.
So I can guess the issue is triggered by the switch "plugin"
Hope this can help to shade some lights to this nasty bug!

psy0rz, 2017-11-25:
Hmm maybe it has to dk with the way RUN_EVERY_SECOND is handled. Probably a too small variable type. Definitly will need to fix this in 2.0.

psy0rz, 2017-11-25:
Millis (): Returns the number of milliseconds since the Arduino board began running the current program. This number will overflow (go back to zero), after approximately 50 days.

So thats the problem. Wonder if this is a long standing issue or if i screwed it up myself. ;)

psy0rz, 2017-11-25:
Unsigned long is 32 bits, which is 49 days in milliiseconds.

However it should wrap around automaticly:

Unsigned long a=0xffffffff;

A=a+1000;

A should be 1000 now, i would expect? Will try this later (on my phone now)

TD-er, 2017-11-25:
I guess it is not in the Millis() code, but rather where it is used.
Perhaps some timer somewhere what tries to see if enough time has passed since the last.
Something like
```
if (newtime > oldtime + 60) {
  oldtime = newtime;
}
```
Will never finish and never recover.

psy0rz, 2017-11-25:
It does. However it should wrap around.

Currently:

Timer1s=timer1s+1000;

This is probably undefined behaviour which might have been fine with previous compilers.

Should be changef to:

Timer1s=timer1s+1000u;

To force unsigned behaviour and insure correct wrapping around.

psy0rz, 2017-11-25:
Confirmed that this code was the same in the pre v2.0 versions. So it was either a long standing bug or undefined behaviour which happened  to work correctly before. (Due to different compiler/optimisation settings)

psy0rz, 2017-11-25:
Also explains why the webgui and gpio stuff is fine.

psy0rz, 2017-11-25:
Just tried this: 
```
  unsigned long a=ULONG_MAX;
  log = F("ULONG_MAX=");
  log += a;
  addLog(LOG_LEVEL_INFO, log);

  a=a+1000;

  log = F("+1000=");
  log += a;
  addLog(LOG_LEVEL_INFO, log);
```

Output:
```
ULONG_MAX=4294967295
+1000=999
```

So this is the output what would be expected. I compiled this with platformio.

Did you guys compile with platformio or Aduino?

papperone, 2017-11-25:
I can confirm it's a bug since R148 as I remember happening since more than 1 year with my modules but not being too annoying I've never payed too much attention on the pattern; as well I do compile with arduino IDE but I tend to use only GitHub BIN on my real modules and self-compiled only during testing

TD-er, 2017-11-25:
@psy0rz 

> It does. However it should wrap around.

I guess it does wrap, and it should.
What I meant was that its usage is probably incorrect and that would explain why some parts keep working (or never use the time) and some other parts don't.

Other thing what I was thinking of... What if the code for the timer was internally using a bigger int, but reading only a 32-bit register from the CPU? Then you would get weird behavior and it's not an overflow.

papperone, 2017-11-25:
but assuming it is wrapping correctly, shouldn't the code check when the wrap happens and adjust the variables?

TD-er, 2017-11-25:
I think there should be a single function in one place to check if enough time has passed.
Time related functions always seem very simple, but they always are more tricky than expected.
So I think there should be something like a time class in our code. But fixing it like that may be a little too much for the 2.0 release. So perhaps -when this seems fixed- add another issue to fix it properly for future releases?

TD-er, 2017-11-25:
In Controller.ino there is some tricky code:
```
    uint16_t dif = millis() - lastSend;
    if (dif < Settings.MessageDelay)
```

And also in Misc.ino:
```
void delayBackground(unsigned long delay)
{
  unsigned long timer = millis() + delay;
  while (millis() < timer)
    backgroundtasks();
}
```
This may never get out of the while loop (for the next 49.7 days, or longer ;) )

And this (also in Misc.ino):
```
void setTime(unsigned long t) {
  sysTime = (uint32_t)t;
  nextSyncTime = (uint32_t)t + syncInterval;
  prevMillis = millis();  // restart counting from now (thanks to Korman for this fix)
}

unsigned long now() {
  // calculate number of seconds passed since last call to now()
  while (millis() - prevMillis >= 1000) {
    // millis() and prevMillis are both unsigned ints thus the subtraction will always be the absolute value of the difference
    sysTime++;
    prevMillis += 1000;
  }
  if (nextSyncTime <= sysTime) {
    unsigned long  t = getNtpTime();
    if (t != 0) {
      if (Settings.DST)
        t += SECS_PER_HOUR; // add one hour if DST active
      setTime(t);
    } else {
      nextSyncTime = sysTime + syncInterval;
    }
  }
  breakTime(sysTime, tm);
  return (unsigned long)sysTime;
}
```
nextSyncTime, sysTime and prevMillis are (re)set in setTime(), but that may not be called when nextSyncTime > sysTime.

TD-er, 2017-12-03:
Hmm, found this in Misc.ino:

```
/********************************************************************************************\
  Unsigned long Timer timeOut check
  \*********************************************************************************************/

boolean timeOut(unsigned long timer)
{
  // This routine solves the 49 day bug without the need for separate start time and duration
  //   that would need two 32 bit variables if duration is not static
  // It limits the maximum delay to 24.9 days.

  unsigned long now = millis();
  //XXX: fix me, something fishy going on here, this << operator is in the wrong place or parenthesis are wrong
  if (((now >= timer) && ((now - timer) < 1u << 31))  || ((timer >= now) && (timer - now > 1u << 31)))
    return true;

  return false;
}
```

Will try to do some digging today.

TD-er, 2017-12-03:
I am still processing all related issues with timeouts and time diff computations.
Currently I am at 19 files changed and some core parts are affected.
So perhaps it should be considered to test this in some kind of 2.0.1 branch, since it will affect lots of parts of the code base.

TD-er, 2017-12-03:
I have tested the timeDiff function and have been running with this code for a few hours.
Also with artificial uptimes of around 2^32 seconds and nothing seemed to crash.
To test, add an offset of something close to 2^32 to both arguments of timeDiff in the function timePassedSince.
For example:
```
long timePassedSince(unsigned long timestamp) {
  unsigned long offset=4294900000u;
  return timeDiff(timestamp + offset, millis() + offset);
}
```

Please have a look at all the locations in the code where these functions are being used, to see whether I made some judgement error on using the code.

psy0rz, 2017-12-04:
@TD-er i've added the comment that says it fishy :)

thanks for trying to fix it and testing it. i think this 49 days bug will be unacceptable in v2.0, so i'll merge your changes after reviewing them.

TD-er, 2017-12-04:
Please note there are still other possible issues which may give similar behavior.
At least all these "little features" I found are taken care of now :)
So at least it should make reproduction of this bug a lot harder now ;)

Grovkillen, 2017-12-19:
We need to test this asap and close the issue if it is resolved.

psy0rz, 2018-01-13:
this should be fixed, testing is difficult since we have to wait 50 days to really test.

We could adjust the timingfunctions with an offset of ~50 days to test this partially.

closed 2018-01-13 by psy0rz

psy0rz, 2018-01-13:
to test i tried replacing all millis() by millisoffset() and added an offset of ULONG_MAX-60

everything seems to keep working. :)

TD-er, 2018-02-19:
My dust sensor outside now is 14232704 msec past the 2^32 msec uptime and it is still working fine.
Just have to test for another day, to allow for some clock drift in freezing temperatures.