letscontrolit/ESPEasy #595: Importing old config into v2.0

psy0rz, 2017-11-20:
Importing an old config doesn't do any error checking currently, resulting into weird bugreports.

1. Create a check to prevent importing a config file of the wrong version/size.

2. Find out if its feasible to convert the old config to the new format. (perhaps just padding with zeros @mvdbro ?)

Grovkillen, 2017-12-19:
Is the old config file also binary or text based? If the later we could make the conversion semi-automatic with a tool?

hexenmeister, 2017-12-20:
Unfortunately, binary. I started to write a parser (https://github.com/hexenmeister/ESPEasy_Config_Loader) (to translate the binaries in text format), but I have no time for that at the moment.

mvdbro, 2017-12-20:
If you look at the code, it should be protected from loading a config file from a different Version.
The Version value should be incremented whenever the internal data structure is changed in such a way that makes things incompatible. (Maybe we forgot to increase the Version ?)

This is were the magic should happen (from the webserver.ino)

// first data block, if this is the config file, check PID/Version
    if (upload.totalSize == 0)
    {
      if (strcasecmp(upload.filename.c_str(), FILE_CONFIG) == 0)
      {
        struct TempStruct {
          unsigned long PID;
          int Version;
        } Temp;
        for (unsigned int x = 0; x < sizeof(struct TempStruct); x++)
        {
          byte b = upload.buf[x];
          memcpy((byte*)&Temp + x, &b, 1);
        }
        if (Temp.Version == VERSION && Temp.PID == ESP_PROJECT_PID)
          valid = true;
      }

mvdbro, 2017-12-20:
As the config file is just a binary flash memory dump, it's difficult to patch from one version to another.

Grovkillen, 2018-01-02:
I could make an web-scraper for people to scrape old setups and convert them to new setups. If it's not working to make this a logical converter?

psy0rz, 2018-01-02:
it is possible to do this, we just have to make a parser/generator in python for example, that understands the old and new structs. but its a bit tricky :)

Grovkillen, 2018-01-17:
As a totally "analog" way of doing backup of settings I created this:
https://github.com/Grovkillen/ESP_Easy_Webdumper

The aim was for me to keep up with the constant change in the web GUI and make screenshots of all the pages but I made it a bit more versatile and it's now batch:able using command line switches for mass-download. Only Windows though... :smirk:

M0ebiu5, 2018-03-14:
Maybe the converter could be a javascript in the browser, which runs before upload?

tonhuisman, 2023-09-03:
A check is in place when uploading a `config.dat` file, so accidentally uploading an older version settings is not possible. That should resolve this issue, so it can be closed.

closed 2023-09-04 by TD-er